microsoft/AirSim #350: Unable to make PX4 1.5.5 on Mac

AndrewEj, 2017-07-18:
My setup :
System: MacOS
i5 CPU @2.00 GHz
8GB RAM
SSD harddrive
RC : No RC
Flight controller : SITL
UE 4 Version: 4.15.3

When I use PX 1.5.5, I got this compile error. 
[image]

When I use the latest version of the PX, I can set up everything. 
[image]
[image]
[image]

However, I cannot control my drone. 

Please help.

sytelus, 2017-07-19:
I think few people have reported this issue. The root cause is that AirSim is not able to connect to SITL. If you are familiar with QGroundControl, first try it out if you can connect with it.

AndrewEj, 2017-07-19:
I think QGroundControl has already connected, but I don't know why there is no reaction inside UE4?  One stupid question is do you need a joystick or not ?

sytelus, 2017-07-19:
Joystick is not required. I am wondering if this is related to crash I see on Linux. I will try to debug this tomorrow.

AndrewEj, 2017-07-19:
Okay !  I don't know what's the problem with PX4 1.5.5;

sifengyang, 2017-07-24:
@AndrewEj  Have you solve the problem？ my problem is "not receiving any message from HIL, please restart your HIL node and try again".

AndrewEj, 2017-07-25:
@sifengyang not yet.. Yeah I have the same problem..

sytelus, 2017-08-14:
First thing to try is make sure your PX4 setup is working with QGroundControl. That is, you should be able to arm, take off etc from QGC. If you can't then you have problem with PX4 setup and you probably want to report this to PX4 forums or Github issues at https://github.com/PX4/Firmware/issues.

closed 2017-08-14 by sytelus