microsoft/AirSim #351: can not build AirSim on Ubuntu 14.04

sifengyang, 2017-07-18:
I am working on Ubuntu 14.04, trying to get the AirSim installed. I have followed the instructions from https://github.com/Microsoft/AirSim/blob/master/docs/linux_build.md

The installation of clang is successful, however when it comes to installing libc++ library, I got a problem 
```
/opt/AirSim/cmake /opt/AirSim/cmake
CMake Error: The source directory "/opt/AirSim/llvm-source" does not exist.
```
by using the ./getlibcxx.sh command.

I manually added the llvm-source folder, but then it says Makefile.config doesn't exist 

Anyone got the issue?

Mayankm96, 2017-07-18:
If you may open the `getlibcxx.sh` file, you would notice that the cloning llvm-source lines have been commented out. I am uncertain why it has been done, but you may just remove them from comment and run the script again.

sifengyang, 2017-07-19:
Thanks, I get AirSim installed.
After the installation, how could I link the unrealengine with AirSim, there is no further instruction on this, the only setup video is  based on Windows.

Mayankm96, 2017-07-19:
Check [#348](https://github.com/Microsoft/AirSim/issues/348). 

Welcome to the club! :D

sifengyang, 2017-07-19:
@Mayankm96 I tried built the AirSim again,  but still showing the same problem. 
`' The CLR exception details may provide more information:  System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.Exception: EIGEN_ROOT is not defined, or points to a non-existant directory, please set this environment variable.  See readme: https://github.com/Microsoft/AirSim/blob/master/docs/install_eigen.md`
'
`printenv EIGEN_ROOT ` can clearly show my eigen path '' at /opt/AirSim/eigen/eigen/eigen3, but why this problem is showing again

sytelus, 2017-07-19:
I think you don't have eigen folder under AirSim folder for whatever reason. The message is misleading in that sense. Today I spent lot of time revamping our build scripts and cmake files. The new build script doesn't require you to install libc++ and also it handles eigen installation correctly. Instructions : https://github.com/Microsoft/AirSim/blob/master/docs/linux_build.md. I haven't got chance to test everything but you might want to try it out.

sifengyang, 2017-07-22:
@sytelus thanks for the update, I rebuilt the source but still got a error

```
CMake Error at /usr/local/share/cmake-3.8/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find Eigen3 (missing: EIGEN3_INCLUDE_DIR EIGEN3_VERSION_OK)
  (Required is at least version "3")
Call Stack (most recent call first):
  /usr/local/share/cmake-3.8/Modules/FindPackageHandleStandardArgs.cmake:377 (_FPHSA_FAILURE_MESSAGE)
  cmake-modules/FindEigen3.cmake:86 (find_package_handle_standard_args)
  cmake-modules/CommonSetup.cmake:9 (find_package)
  AirLib/CMakeLists.txt:8 (IncludeEigen)
```
Tried solve the problem, i manually set EIGEN

```
echo export EIGEN_ROOT=$(pwd)/llvm-source/eigen >> ~/.bashrc
source ~/.bashrc
```
I can see EIGEN_ROOT=/opt/AirSim/llvm-source/eigen through printenv

but still build.sh causes this problem.

sytelus, 2017-07-22:
I would suggest. 
1. Remove anything in .bashrc to set EIGEN_ROOT
2. `unset EIGEN_ROOT`
3. `git clean -fdx`
4. `git pull`
5. `./setup.sh`
6. `./build.sh`

Please let me know if these steps work.

helxsz, 2017-07-22:
The new approach works for me finally.

sifengyang, 2017-07-22:
@sytelus I installed twice, the first time I did the git pull, but still have EIGEN_ROOT left in the environmental variable, so unsuccessfully installed. The second time I unset EIGEN_ROOT, so it worked perfectly.  

Thanks a lot!

closed 2017-07-22 by sifengyang