microsoft/AirSim #352: New C++ API image issues

ShukuiZhang, 2017-07-18:
I'm now implementing the new C API for to get image and poses, it gives me weird results in the following two senses:
1. Scene image is really dark. The screenshot looks natural, e.g. in following screenshot
[image] but the scene image I got looks like
[image] Previously, the scene image looks okay, this happened when I implement the new api.
I have the common setting file.
```
{
  "LocalHostIp": "127.0.0.1",
  "Pixhawk": {
    "LogViewerHostIp": "127.0.0.1",
    "LogViewerPort": 14389,
    "OffboardCompID": 1,
    "OffboardSysID": 134,
    "QgcHostIp": "127.0.0.1",
    "QgcPort": 14550,
    "SerialBaudRate": 115200,
    "SerialPort": "*",
    "SimCompID": 42,
    "SimSysID": 142,
    "SitlIp": "127.0.0.1",
    "SitlPort": 14556,
    "UdpIp": "127.0.0.1",
    "UdpPort": 14560,
    "UseSerial": true,
    "VehicleCompID": 1,
    "VehicleSysID": 135
  },
  "SceneCaptureSettings" : {
    "Width": 640,
	  "Height": 480,
    "FOV_Degrees": 90,
    "AutoExposureSpeed": 1000,
    "MotionBlurAmount": 0
  },  
  "DepthCaptureSettings" : {
    "Width": 640,
	  "Height": 480,
    "FOV_Degrees": 90,
    "AutoExposureSpeed": 1000,
    "MotionBlurAmount": 0
  },
  "RpcEnabled": true
}
```
2. How to use the depth image? E.g. I saved the depth_raw [line](https://github.com/Microsoft/AirSim/blob/master/Examples/StereoImageGenerator.hpp#L165) directly to an png file and then used as a depth map to work out the 3D point cloud. I know to convert the ray depth to plane depth to work on, but still gives me really weird results. What is the unit of the depth values?

sytelus, 2017-07-19:
That is very strange. Can you please try out latest code (I made few changes today). One thing I observed was that scene image sometime doesn't adjust when you move from bright area to dark area (for example, moving in shadow). However this doesn't seem to be your case. One thing I would suggest is to experiment with various settings in PIPCamera.h file. Just remove settings from your setting.json so all settings would whatever default you apply in PIPCamera.h. Please let me know if any combination of settings give you good result.

ShukuiZhang, 2017-07-19:
Hi @sytelus . I did what you suggested above. I reset the settings in settings.json file and then tried to modify the parameters in PIPCamera.h, both for previous code and latest commit. But I get no luck, the weird part is that I can merely see any changes in image illumination even I made big changes to the parameters. (Make no sense, hahh?)

On the other hand, I have tried different things. E.g. I noticed that if I make captured image NOT compressed (e.g. ImageRequest(1, ImageType_::Scene, false, true)) and save such image, the illumination is more reasonable. But that is very slow. So do you think function [saveImageToFile()](https://github.com/Microsoft/AirSim/blob/master/Examples/StereoImageGenerator.hpp#L193) decompression might be a problem? Thanks.

sytelus, 2017-07-20:
That's good hint. Would you be able to post the screenshot of uncompressed version? Is it just "reasonable" or what you would expect? Also, how are you saving the uncompressed image? I think uncompressed image isn't in png format.

ShukuiZhang, 2017-07-20:
This is the image after applying the StereoImageGenerator.hpp
[image]
This is the uncompressed version:
[image]
To me it looks okay since I noticed it looks similar to what appears in the scene subwindow. 

It seems tedious to save the uncompressed version. Firstly, I save the uncompressed data to a csv file in c code. I then did some postprocessing in Matlab to transform the csv data back to 3D array and then save as an image. This is merely for a test purpose.
BTW, I noticed that even if I removed PIPcamera.h and PIPcamera.cpp from the Plugins folder in my project, I can still run the project and capture images with no problem. Does it mean the settings in PIPcamera.h actually overwrite some global settings somewhere?

ShukuiZhang, 2017-07-21:
Today I create a new project using another module, now the picture is too bright... Similar to [issue](https://github.com/Microsoft/AirSim/issues/359)

sytelus, 2017-07-22:
Could you please share your matlab code? May be I can try that out on my m/c.

PIPCamera* is absolutely required. If you clean the solution and try to build it, it won't even compile.

Which env are you using? I'm seeing bit darker images but not super dark images that you are seeing. May be I can try out that environment?

I've added more settings for image capture so you can play around:
```

  "SceneCaptureSettings" : {
    "Width": 256,
	  "Height": 144,
    "FOV_Degrees": 90,
    "AutoExposureSpeed": 100,
    "AutoExposureBias": 0,
    "AutoExposureMaxBrightness": 0.64f,
    "AutoExposureMinBrightness": 0.03f,
    "MotionBlurAmount": 0,
    "TargetGamma": 1.0f
  }
```

More info here: https://github.com/Microsoft/AirSim/blob/master/docs/settings.md

Please let us know which settings works for you so I can add that info in docs.

ShukuiZhang, 2017-07-22:
Here is the code to save as csv in c code:
```
static void saveImageToSceneCSV(const std::vector<uint8_t>& image_data, const std::string& file_name)
	{
		std::ofstream filestream(file_name, std::ios::binary);

		// do not save the transparent channel
		for (int i = 0; i < image_data.size(); ++i)
			if ((i+1) % 4 != 0)
				filestream << std::to_string(double(image_data[i])) << ",";
		filestream << "\n";
		filestream.close();

	}
```
The following is the matlab code that for postprocessing:
```
scene1  = 'C:\\Users\\shzhang\Documents\AirSim\data_2\leftcsv_000001.csv';

s1_csv = csvread(scene1);

width = 640;
height = 480;
for i= 1: height
    for j =1: width
        for k = 1:3 
            scene1_img(i,j,k) = s1_csv(3* ((i-1) * width + (j-1)) + k);
        end
    end
end

scene1_img = mat2gray(scene1_img);

figure(1)
imshow(scene1_img)
```
The env is UE4 sun temple. I will definitely try your new code. Thanks!

xuhao1, 2018-06-08:
Is this issue solved?
When I using SimGetImage to get scene image from unreal environment, the environment is also very dark
[image]

fernandodamasio, 2018-10-23:
Hi guys,

I´m having a problem with the RECORD function to capture images from the simulator.

When I press R to start recording, everything looks ok. But when I check the images generated in the DOCUMENTS/AIRSIM folder, they are all very "white". Examples:

[image]

[image]

However, the images generated REALTIME are OK.

[image]

I tried different solutions like tunning the `"CameraDefaults": { "CaptureSettings": [` parameters but no success.

I also tried to exclude the foliage I was using because It was generating a non-critical error after BUILD LIGHTING. Again no success.

I´m running the simulation with EPIC QUALITY with an NVIDIA GTX 680 GPU.

I´m not using any kind of FOG.

Any insights here?

Best regards,
Fernando Damasio

fernandodamasio, 2018-10-25:
Hi again guys.

I just solved the problem. Looks like I was missing the Directional Light for the sky in the Level. I don't know if this was the problem, as I had to do the level again from scratch, but I believe this was the cause.

NextSim, 2019-01-04:
Did the directional light fix this?

Can anyone provide steps to check if the directional light for the sky is there?  For people new to unreal it is not easy to find.

Thanks

madratman, 2019-01-24:
I'd suggest playing with the targetgamma setting. 
Here's a sample settings.json:

```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 1080,
        "Height": 720,
        "FOV_Degrees": 120,
        "TargetGamma": 3.0
      }
    ]
}
}
```

As mentioned in [this comment](https://github.com/Microsoft/AirSim/issues/1033#issuecomment-385999993), the auto-exposure settings need some work to get enabled. We'll update the docs.

michael-fonder, 2019-02-14:
@madratman The solution I proposed does not work with the precompiled releases of Airsim, since it requires to make modifications to an asset. I'll try  to find time in the following days to make a proper pull request to fix this issue once and for all.

madratman, 2019-02-19:
That would be really appreciated @michael-fonder. Please let us know if you need any help with that

closed 2020-03-12 by ironclownfish