microsoft/AirSim #353: DroneCommand Code Structure

fmzhu, 2017-07-19:
Hello, could anyone explain that, how  the "Control Command"  are sent to the PX4,  since the sensordata are sent through the "sendHILSensor " based on MAVLinkmessage, so how the "Control Command" works, is it a MAVLinkmessage too? Thank you very much!

sytelus, 2017-07-19:
Can you elaborate what you mean by Control Command? All communication with PX4 are through MavLink messages.

fmzhu, 2017-07-19:
@sytelus I mean the DroneShell Command, how are the Command sent to the PX4, because I want to make Networkfaultinjection between AirSim and PX4. Thank you very much for your attention!

sytelus, 2017-07-19:
DroneShell talks to AirSim via RPC. The RPC call on AirSim side is handled by DroneControllerBase which in case of PX4 will translate the RPC command to MavLink command.

fmzhu, 2017-07-20:
@sytelus thank you very much for your help! so would you please explain how are the RPC commands translated to the Mavlink commands. Thank you very much for your attention!

lovettchris, 2017-07-20:
A picture helps:
[image]

fmzhu, 2017-07-21:
@lovettchris Thank you very much for your help, that's will be very helpful!

closed 2017-07-22 by sytelus