microsoft/AirSim #354: Unreal Engine crashed when clicking play button (git checkout to the latest)

ZongliangLee, 2017-07-19:
Unreal Engine crashed both on :
0c87453ee102002dd4847a985758b4d1c8819774  (the latest)
0aa7f3b06b7f5848d07230a1c15efc3e1a83ad8f

I've got the error message below:

[2017.07.19-07.45.15:473][823]LogLinux: === Critical error: ===
Unhandled Exception: SIGABRT: abort() called

[2017.07.19-07.45.15:473][823]LogLinux: Fatal error!

[Callstack]  00  0x00007f43b2d02cff  FLinuxPlatformStackWalk::CaptureStackBackTrace(unsigned long long*, unsigned int, void*)
[Callstack]  01  0x00007f43b2b99095  FGenericPlatformStackWalk::StackWalkAndDump(char*, unsigned long, int, void*)
[Callstack]  02  0x00007f43b2ca8151  FLinuxCrashContext::CaptureStackTrace()
[Callstack]  03  0x00007f43a781e3e0  CommonLinuxCrashHandler(FGenericCrashContext const&)
[Callstack]  04  0x00007f43b2caa022  PlatformCrashHandler(int, siginfo_t*, void*)
[Callstack]  05  0x00007f43b8aec390  /lib/x86_64-linux-gnu/libpthread.so.0(+0x11390) [0x7f43b8aec390]
[Callstack]  06  0x00007f43a68eb428  /lib/x86_64-linux-gnu/libc.so.6(gsignal+0x38) [0x7f43a68eb428]
[Callstack]  07  0x00007f43a68ed02a  /lib/x86_64-linux-gnu/libc.so.6(abort+0x16a) [0x7f43a68ed02a]
[Callstack]  08  0x00007f43a692d7ea  /lib/x86_64-linux-gnu/libc.so.6(+0x777ea) [0x7f43a692d7ea]
[Callstack]  09  0x00007f43a693637a  /lib/x86_64-linux-gnu/libc.so.6(+0x8037a) [0x7f43a693637a]
[Callstack]  10  0x00007f43a693a53c  /lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c) [0x7f43a693a53c]
[Callstack]  11  0x00007f42a429e359  AAirSimGameMode::initializeSettings()
[Callstack]  12  0x00007f42a429db39  AAirSimGameMode::StartPlay()
[Callstack]  13  0x00007f43af698ca9  UWorld::BeginPlay()
[Callstack]  14  0x00007f43aeafacf5  UGameInstance::StartPlayInEditorGameInstance(ULocalPlayer*, FGameInstancePIEParameters const&)
[Callstack]  15  0x00007f43a931adb6  UEditorEngine::CreatePIEGameInstance(int, bool, bool, bool, bool, float)
[Callstack]  16  0x00007f43a9313126  UEditorEngine::PlayInEditor(UWorld*, bool)
[Callstack]  17  0x00007f43a9303d93  UEditorEngine::StartQueuedPlayMapRequest()
[Callstack]  18  0x00007f43a8cf5f76  UEditorEngine::Tick(float, bool)
[Callstack]  19  0x00007f43a96c8629  UUnrealEdEngine::Tick(float, bool)
[Callstack]  20  0x0000000000423a30  FEngineLoop::Tick() [Runtime/Core/Public/Stats/Stats2.h, line 1686]
[Callstack]  21  0x000000000042b2a3  GuardedMain(wchar_t const*) [/test/testbed/home/test/Desktop/UnrealEngine/UnrealEngine/Engine/Source/Runtime/Launch/Private/Launch.cpp, line 62]
[Callstack]  22  0x00007f43a781f16a  CommonLinuxMain(int, char**, int (*)(wchar_t const*))
[Callstack]  23  0x00007f43a68d6830  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7f43a68d6830]
[Callstack]  24  0x0000000000416669  /test/testbed/home/test/Desktop/UnrealEngine/UnrealEngine/Engine/Binaries/Linux/UE4Editor(_start+0x29) [0x416669]

[2017.07.19-07.45.15:483][823]LogExit: Executing StaticShutdownAfterError
Malloc Size=274008 LargeMemoryPoolOffset=470680 
Pooling OS allocations (pool size: 1408 MB, maximum allocations: 9502).
Using binned2.
LogPlatformFile: Not using cached read wrapper
LogInit:Display: RandInit(-804334270) SRandInit(-804334270).
LogTaskGraph: Started task graph with 4 named threads and 7 total threads with 1 sets of task threads.
LogInit: Build: ++UE4+Release-4.16-CL-0
LogInit: Engine Version: 4.16.2-0+++UE4+Release-4.16
LogInit: Compatible Engine Version: 4.16.0-0+++UE4+Release-4.16
LogInit: Net CL: 0
LogInit: Compiled (64-bit): Jul 18 2017 17:03:39
LogInit: Compiled with Clang: 3.8.0 (tags/RELEASE_380/final)
LogInit: Build Configuration: Shipping
LogInit: Branch Name: ++UE4+Release-4.16
LogInit: Command line:  -Abslog=/home/bob/Unreal Projects/airsim0719_5/Saved/Logs/airsim0719_5-CRC.log /home/bob/Unreal Projects/airsim0719_5/Saved/Crashes/crashinfo-airsim0719_5-pid-14179-208A69130F070F13002D01E4BCEEAEC0/
LogInit: Base directory: /test/testbed/home/test/Desktop/UnrealEngine/UnrealEngine/Engine/Binaries/Linux/
LogInit: Installed Engine Build: 1
LogInit: Presizing for max 100000 objects, including 0 objects not considered by GC, pre-allocating 0 bytes for permanent pool.
LogInit: Object subsystem initialized
[2017.07.19-07.45.15:562][  0]LogInit: Linux hardware info:
[2017.07.19-07.45.15:562][  0]LogInit:  - we are the first instance of this executable
[2017.07.19-07.45.15:562][  0]LogInit:  - this process' id (pid) is 14321, parent process' id (ppid) is 14179
[2017.07.19-07.45.15:562][  0]LogInit:  - we are not running under debugger
[2017.07.19-07.45.15:562][  0]LogInit:  - machine network name is 'bob-Z97-D3H'
[2017.07.19-07.45.15:562][  0]LogInit:  - user name is 'bob' (bob)
[2017.07.19-07.45.15:562][  0]LogInit:  - we're logged in locally
[2017.07.19-07.45.15:562][  0]LogInit:  - we're running with rendering
[2017.07.19-07.45.15:562][  0]LogInit:  - CPU: GenuineIntel 'Intel(R) Core(TM) i7-4790 CPU @ 3.60GHz' (signature: 0x306C3)
[2017.07.19-07.45.15:562][  0]LogInit:  - Number of physical cores available for the process: 4
[2017.07.19-07.45.15:562][  0]LogInit:  - Number of logical cores available for the process: 8
[2017.07.19-07.45.15:562][  0]LogInit:  - Cache line size: 64
[2017.07.19-07.45.15:562][  0]LogInit:  - Memory allocator used: binned2
[2017.07.19-07.45.15:562][  0]LogLinux: Skipped benchmarking clocks because the engine is running in a standalone program mode - CLOCK_REALTIME will be used.
[2017.07.19-07.45.15:562][  0]LogInit: Linux-specific commandline switches:
[2017.07.19-07.45.15:562][  0]LogInit:  -nodwarf (currently OFF): suppress parsing of DWARF debug info (callstacks will be generated faster, but won't have line numbers)
[2017.07.19-07.45.15:562][  0]LogInit:  -ansimalloc - use malloc()/free() from libc (useful for tools like valgrind and electric fence)
[2017.07.19-07.45.15:562][  0]LogInit:  -jemalloc - use jemalloc for all memory allocation
[2017.07.19-07.45.15:562][  0]LogInit:  -binnedmalloc - use binned malloc  for all memory allocation
[2017.07.19-07.45.15:562][  0]LogInit:  -httpproxy=ADDRESS:PORT - redirects HTTP requests to a proxy (only supported if compiled with libcurl)
[2017.07.19-07.45.15:562][  0]LogInit:  -reuseconn - allow libcurl to reuse HTTP connections (only matters if compiled with libcurl)
[2017.07.19-07.45.15:562][  0]LogInit:  -virtmemkb=NUMBER - sets process virtual memory (address space) limit (overrides VirtualMemoryLimitInKB value from .ini)
[2017.07.19-07.45.15:562][  0]LogInit:  - Physical RAM available (not considering process quota): 16 GB (15882 MB, 16264120 KB, 16654458880 bytes)
[2017.07.19-07.45.15:563][  0]LogUObjectArray: 146 objects as part of root set at end of initial load.
[2017.07.19-07.45.15:563][  0]LogUObjectAllocator: 24112 out of 0 bytes used by permanent object pool.
[2017.07.19-07.45.15:563][  0]LogUObjectArray: CloseDisregardForGC: 0/0 objects in disregard for GC pool
[2017.07.19-07.45.15:563][  0]LogInit: Using OS detected language ().
[2017.07.19-07.45.15:563][  0]LogInit: Using OS detected locale ().
[2017.07.19-07.45.15:563][  0]LogTextLocalizationManager: No localization for '' exists, so 'en' will be used for the language.
[2017.07.19-07.45.15:563][  0]LogTextLocalizationManager: No localization for '' exists, so 'en' will be used for the locale.
[2017.07.19-07.45.15:581][  0]CrashReportClientLog: CrashReportClientVersion=1.0
[2017.07.19-07.45.15:581][  0]CrashReportClientLog: CrashReportReceiver disabled
[2017.07.19-07.45.15:581][  0]CrashReportClientLog: DataRouterUrl: https://datarouter.ol.epicgames.com/datarouter/api/v1/public/data
[2017.07.19-07.45.15:581][  0]LogExit: Preparing to exit.
[2017.07.19-07.45.15:581][  0]LogObj: Freed 0b from 0 cluster array pools.
[2017.07.19-07.45.15:581][  0]LogExit: Object subsystem successfully closed.
[2017.07.19-07.45.15:582][  0]LogModuleManager: Shutting down and abandoning module CoreUObject (2)
[2017.07.19-07.45.15:582][  0]LogExit: Exiting.
Engine crash handling finished; re-raising signal 6 for the default handler. Good bye.

sytelus, 2017-07-19:
I am investigating this and can replicate on my machine as well. Note that this happens when trying to use Pixhawk. The Computer Vision mode seems to be ok...

Mayankm96, 2017-07-24:
Any update on this @sytelus ?

sytelus, 2017-07-27:
During last week I've fixed various Linux issues, one of the core one (above) turned out to be low level bug in Unreal engine. The setup and build for Linux is all revamped so you might want to try out the latest. Please (see [doc](https://github.com/Microsoft/AirSim/blob/master/docs/linux_build.md)) for setup instructions and known issues in Linux. The PX4 1.5.5 SITL isn't as stable but you can give it try as well.

closed 2017-07-27 by sytelus