microsoft/AirSim #355: Missing C++ dependencies in Ubuntu 16.04

curantil, 2017-07-19:
Following the install instructions for Linux I got errors about missing libc++ and libc++abi when running ./build.sh. 
```
clang: warning: argument unused during compilation: '-nostdinc++' [-Wunused-command-line-argument]
/usr/bin/ld: cannot find -lc++
/usr/bin/ld: cannot find -lc++abi
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

After installing those with ```sudo apt install libc++abi-dev libc++-dev``` I still get the error ```/usr/bin/ld: cannot find -lc++experimental``` for which I do not find how to install it.

sytelus, 2017-07-19:
I'm wondering if setup.sh failed. I've added `set -x` so you can see the error better and also made other fix. Could you please try out the latest? You should see a message at the end of the setup.sh that it was successful.

closed 2017-07-19 by sytelus

jack1986526, 2017-07-20:
I have the same error.