microsoft/AirSim #356: EIGEN_ROOT not set

curantil, 2017-07-19:
Following the installation instructions for Linux.

EIGEN_ROOT is not set after running ./setup.sh so ./build.sh gives an error about it. Manually setting the directory to where the archive was extracted seems to work.

sytelus, 2017-07-19:
I was just able to replicate this issue and made a fix. Please try out the latest build script!

closed 2017-07-19 by sytelus

curantil, 2017-07-20:
I did a pull, ./setup.sh and ./build.sh

The end of the setup.sh seemed successful:
```
  inflating: eigen/eigen-eigen-da9b4e14c255/unsupported/test/splines.cpp  
+ pushd eigen
~/projects/AirSimNative/GitHubSrc/AirSim/llvm-build/eigen ~/projects/AirSimNative/GitHubSrc/AirSim/llvm-build ~/projects/AirSimNative/GitHubSrc/AirSim
+ mv eigen-eigen-da9b4e14c255 eigen3
+ echo 3.3.2
+ popd
+ rm 3.3.2.zip
+ popd
~/projects/AirSimNative/GitHubSrc/AirSim
+ set +x

************************************
AirSim setup completed successfully!
************************************
```

Running build.sh still seems to give the same problems:
```
$ ./build.sh 
~/projects/AirSimNative/GitHubSrc/AirSim ~/projects/AirSimNative/GitHubSrc/AirSim
+ git submodule update --init --recursive
+ [[ -d ./llvm-build/output/lib ]]
+ export CC=/usr/bin/clang-3.9
+ CC=/usr/bin/clang-3.9
+ export CXX=/usr/bin/clang++-3.9
+ CXX=/usr/bin/clang++-3.9
+ [[ -z '' ]]
+ echo 'EIGEN_ROOT variable is not set'
EIGEN_ROOT variable is not set
+ [[ ! -d eigen ]]
+ echo 'eigen is not installed. Please run setup.sh first.'
eigen is not installed. Please run setup.sh first.
+ exit 1
```

shiran8338, 2020-07-19:
Hi! Have you figured this out? I have the same problem right now.