microsoft/AirSim #357: RPC client crash in ComputerVision mode

gawela, 2017-07-19:
Hello all,

I am trying to set up programmatic control of the drone in ComputerVision mode (no px4 running). However, whenever the RPC client tries to connect to the simulation it crashes due to the `SimpleFlightController::getGpsLocation()` [here](https://github.com/Microsoft/AirSim/blob/master/AirLib/include/controllers/simple_flight/SimpleFlightDroneController.hpp#L194).

It appears that it cannot receive the gps location. Is there a setting I may be overlooking? 
It does not crash when using RosFlight instead of SimpleFlight, but then the commands don't get executed by the Drone. Any hints would be greatly appreciated.

sytelus, 2017-07-20:
What's in your settings.json? Which OS are you using?

gawela, 2017-07-20:
Thank you for the quick response.
I am on Windows 10 and the stock .json:

`{
  "LocalHostIp": "127.0.0.1",
  "RecordUIVisible": true,
  "LogMessagesVisible": true,
  "ViewMode": "FlyWithMe",
  "FpvVehicleName": "SimpleFlight",
  "UsageScenario": "ComputerVision",
  "RpcEnabled": true,
  "RosFlight": {
    "RemoteControlID": 0
  },
  "SceneCaptureSettings" : {
    "Width": 256,
	  "Height": 144,
    "FOV_Degrees": 90,
    "AutoExposureSpeed": 100,
    "MotionBlurAmount": 0
  },  
  "DepthCaptureSettings" : {
    "Width": 256,
	  "Height": 144,
    "FOV_Degrees": 90,
    "AutoExposureSpeed": 100,
    "MotionBlurAmount": 0
  },  
  "SegCaptureSettings" : {
    "Width": 256,
	  "Height": 144,
    "FOV_Degrees": 90,
    "AutoExposureSpeed": 100,
    "MotionBlurAmount": 0
  },  
  "SimpleFlight": {
    "RemoteControlID": 0
  },
  "Pixhawk": {
    "LogViewerHostIp": "127.0.0.1",
    "LogViewerPort": 14388,
    "OffboardCompID": 1,
    "OffboardSysID": 134,
    "QgcHostIp": "127.0.0.1",
    "QgcPort": 14550,
    "SerialBaudRate": 115200,
    "SerialPort": "*",
    "SimCompID": 42,
    "SimSysID": 142,
    "SitlIp": "127.0.0.1",
    "SitlPort": 14556,
    "UdpIp": "127.0.0.1",
    "UdpPort": 14560,
    "UseSerial": true,
    "VehicleCompID": 1,
    "VehicleSysID": 135
  }
}
`

gawela, 2017-07-20:
I was looking further into the issue and found out that the Plugins were deprecated in my Unreal project, as some functions got new names (non-existing `simSetPose()` was causing the issue), giving me this cryptic behaviour - now it works fine.

Thank you for the help! :)

closed 2017-07-20 by gawela