microsoft/AirSim #358: setup airsim

RichardM678, 2017-07-20:
I am struggling in using build.cmd
when I type the command, it just closes the prompt window. I tried couple of times to reinstall softwares.i am using VS2015. cmake3.90, unreal engine 4.15.3.
I tried a lots of solution on the internet, but it seems like they are not that helpful. 
when I run the command it closed the window instantly, I cannot see what are the specific errors.

RichardM678, 2017-07-20:
I found the error, it says cmake is not found. 
I do have cmake, does the version matter? I have cmake 3.90

RichardM678, 2017-07-20:
okay ,right now I figure it out, the build command runs.
But it says build fails.
There are 5 warnings and 5 errors.
warning is, the file contains a character that cannot be represented in the current code page(936). Save the file in unicode format to prevent data loss.
the error is, error C2220: warning treated as error - no 'object; file generated.

Robin-Wong, 2017-07-20:
Please check the file AirSim\eigen\Eigen3\Eigen\src\Core\arch\CUDA\Half.h, line 16, delete the " on both sides of AS IS.

RichardM678, 2017-07-20:
my 16 line is a comment line, it is 
//''AS IS'' AND ANY EXPRESS OR  IMPLIED WARRANTIES, INCLUDING, BUT NOT
I am using vs2015 to open it, not sure if they generate the same line.
Is it supposed to be inside the code?

RichardM678, 2017-07-20:
nvm I got what your saying
//AS IS AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT
I did this is, it's still not working

Robin-Wong, 2017-07-20:
Please try to run build.cmd in VS2015 x64 Native Tools Command Prompt.

RichardM678, 2017-07-20:
I download the files again, and it works!
thank you so much!!!!!

closed 2017-07-22 by sytelus