microsoft/AirSim #359: How to change the brightness and exposure of the images being recorded ?

titiyagupta, 2017-07-20:
The images recorded look like the view captured in the camera view by pressing 3 : 

[image]

sytelus, 2017-07-22:
Could you please try the latest version? Previously I'd settings that I saw on my m/c but may be that doesn't work. So by default I'm forcing everything to be decided by Unreal engine. I've also added bunch of more options for [settings.json](https://github.com/Microsoft/AirSim/blob/master/docs/settings.md):

```
  "SceneCaptureSettings" : {
    "Width": 256,
	  "Height": 144,
    "FOV_Degrees": 90,
    "AutoExposureSpeed": 100,
    "AutoExposureBias": 0,
    "AutoExposureMaxBrightness": 0.64f,
    "AutoExposureMinBrightness": 0.03f,
    "MotionBlurAmount": 0,
    "TargetGamma": 1.0f
  }
```

Typically you will see washed out picture if TargetGamma is too high or AutoExposureSpeed is too hight (for example, 1000). 

Please let us know what settings works for you so we can update docs.

shristi945, 2017-07-24:
Hi @sytelus  I tried by adding the above setting in settings.json keeping "AutoExposureSpeed" : 35 and "TargetGamma" : 0.1

 "SceneCaptureSettings": {
    "Width": 256,
    "Height": 144,
    "FOV_Degrees": 90,
    "AutoExposureSpeed": 35,
    "AutoExposureBias": 0,
    "AutoExposureMaxBrightness": 0.64,
    "AutoExposureMinBrightness": 0.03,
    "MotionBlurAmount": 0,
    "TargetGamma": 0.1
  },

looks like this is working  and now I am getting a clearer Image.

[image]

Thanks.

shristi945, 2017-07-24:
@sytelus But the image that is being saved in the C:\...\Documents\AirSim\2017-07-24 folder is not clear.
it looks like this:

[image]

Please Help.

helxsz, 2017-07-26:
I 'd like to change the brightness and exposure in the code dynamically,  where can I get a clue for this ?

shristi945, 2017-07-28:
@sytelus  Please give response for the above problem.

Thanks in Advance.

sytelus, 2017-07-29:
I'm not sure... Have to try reproduce this. Few things I would suggest is to (!) try API call and see if you get same result (2) try to get uncompressed image and see if you get same result. If you search GitHub Issue for uncompressed then you will so info on how to do it. I might not get around to take a look at this until 1-2 weeks.

sytelus, 2018-04-12:
This looks like duplicate of https://github.com/Microsoft/AirSim/issues/352. Please continue discussions over that thread.

closed 2018-04-12 by sytelus