microsoft/AirSim #360: Building AirSim on Linux

CalumMurray, 2017-07-20:
I've successfully ran the setup.sh script. libc++ seems to be built correctly, but when I run build.sh, I get the following linking errors when trying to build MavLinkTest:

```
[ 85%] Linking CXX executable ../../output/bin/MavLinkTest
clang-4.0: warning: argument unused during compilation: '-nostdinc++' [-Wunused-command-line-argument]
CMakeFiles/MavLinkTest.dir/~/AirSim/MavLinkCom/MavLinkTest/main.cpp.o: In function `std::experimental::filesystem::v1::path::compare(std::experimental::filesystem::v1::path const&) const':
~/AirSim/llvm-build/output/include/c++/v1/experimental/filesystem:910: undefined reference to `std::experimental::filesystem::v1::path::__compare(char const*) const'
~AirSim/llvm-build/output/include/c++/v1/experimental/filesystem:910: undefined reference to `std::experimental::filesystem::v1::path::__compare(char const*) const'
CMakeFiles/MavLinkTest.dir/~/AirSim/MavLinkCom/MavLinkTest/Commands.cpp.o: In function `std::experimental::filesystem::v1::path::compare(std::experimental::filesystem::v1::path const&) const':
~/AirSim/llvm-build/output/include/c++/v1/experimental/filesystem:910: undefined reference to `std::experimental::filesystem::v1::path::__compare(char const*) const'
CMakeFiles/MavLinkTest.dir/~/AirSim/MavLinkCom/MavLinkTest/Commands.cpp.o: In function `std::experimental::filesystem::v1::directory_iterator::operator*() const':
~/AirSim/llvm-build/output/include/c++/v1/experimental/filesystem:1849: undefined reference to `std::experimental::filesystem::v1::directory_iterator::__deref() const'
clang-4.0: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [MavLinkCom/MavLinkTest/CMakeFiles/MavLinkTest.dir/build.make:148: output/bin/MavLinkTest] Error 1
make[1]: *** [CMakeFiles/Makefile2:272: MavLinkCom/MavLinkTest/CMakeFiles/MavLinkTest.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
```

The only change I made was to use clang 4.0 (as I'm using Unreal 4.16.2).  Would this cause this issue?

ajberchek, 2017-07-20:
#132 Seems to address a similar issue. 

The lack of clang3.9 could be causing the make step in setup.sh to fail to compile all the libc++ files; some of those files handle the experimental filesystem. 

Try running the command in the referenced issue to see if the necessary file is present.

CalumMurray, 2017-07-21:
Thanks for the link to that issue - I did not find that one.

So I had already ran `getlibcxx.sh` and looked like `libc++experimental.a` was built correctly.  However, the `llvm-build/libc++experimental.a` differed from the version in `/usr/lib/libc++experimental.a`.  When I copied the AirSim llvm-build version there manually, it successfully built.  Perhaps there's an issue with an `install libc++experimental.a /usr/lib/libc++experimental.a` command invoked somewhere?

Well, I got it working, thanks for your help.

closed 2017-07-21 by CalumMurray