microsoft/AirSim #361: launch project error

RichardM678, 2017-07-20:
Hey, so after following the video, I generated plugins file and copy to the project file. Then I added the showed in the video.
Here is the code:
{
	"FileVersion": 3,
	"EngineAssociation": "4.15",
	"Category": "",
	"Description": "",
	"Modules": 
	[
		{
			"Name": "Drone",
			"Type": "Runtime",
			"LoadingPhase": "Default",
			"AdditionalDependencies": 
			[
				"AirSim"
			]
			
		}
	],
	"Plugins": 
	[
         {
             "Name": "AirSim",
             "Enabled": true
         }
     ]      
	
	   
}
However, after I run the file, it gave an error:
Running C:/Program Files/Epic Games/UE_4.15/Engine/Binaries/DotNET/UnrealBuildTool.exe  -projectfiles -project="C:/Users/82459/Documents/Unreal Projects/Drone/Drone.uproject" -game -rocket -progress
Discovering modules, targets and source code for project...
ERROR: Couldn't find module rules file for module 'AirSim'.

RichardM678, 2017-07-20:
I tried to redownload everything, there are some new errors pops out.
Discovering modules, targets and source code for project...
ERROR: Unable to instantiate instance of 'AirSim' object type from compiled assembly 'DroneModuleRules, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null'.  Unreal Build Tool creates an instance of your module's 'Rules' object in order to find out about your module's requirements.  The CLR exception details may provide more information:  System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.Exception: EIGEN_ROOT is not defined, or points to a non-existant directory, please set this environment variable.  See readme: https://github.com/Microsoft/AirSim/blob/master/docs/install_eigen.md
   at AirSim.AddEigenDependency() in c:\Users\82459\Documents\Unreal Projects\Drone\Plugins\AirSim\Source\AirSim.Build.cs:line 83
   at AirSim..ctor(TargetInfo Target) in c:\Users\82459\Documents\Unreal Projects\Drone\Plugins\AirSim\Source\AirSim.Build.cs:line 71
   --- End of inner exception stack trace ---
   at System.RuntimeMethodHandle.InvokeMethod(Object target, Object[] arguments, Signature sig, Boolean constructor)
   at System.Reflection.RuntimeConstructorInfo.Invoke(BindingFlags invokeAttr, Binder binder, Object[] parameters, CultureInfo culture)
   at System.RuntimeType.CreateInstanceImpl(BindingFlags bindingAttr, Binder binder, Object[] args, CultureInfo culture, Object[] activationAttributes, StackCrawlMark& stackMark)
   at System.Activator.CreateInstance(Type type, BindingFlags bindingAttr, Binder binder, Object[] args, CultureInfo culture, Object[] activationAttributes)
   at System.Activator.CreateInstance(Type type, Object[] args)
   at UnrealBuildTool.RulesAssembly.CreateModuleRules(String ModuleName, TargetInfo Target, FileReference& ModuleFileName)

lukaszdworako, 2017-07-21:
One error that pops out as obvious would be your EIGEN_ROOT not being properly configured, if you haven't already try following this [tutorial](https://github.com/Microsoft/AirSim/blob/master/docs/install_eigen.md).

RichardM678, 2017-07-21:
How can I properly configure the EIGEN_ROOT.
Here is the instruction, but I don't know how to do this step

But your Unreal Projects will also need to be able to find this location and for that we set an EIGEN_ROOT environment variable. That variable needs to point to the eigen folder

sytelus, 2017-07-22:
Are you using Unreal 4.15 on Windows? Can you please confirm you have Plugins\AirSim folder under your Unreal project? How are you "running" the project? If that is the case you shouldn't get this error. The error means Unreal can't find Build.cs file either AirSim.

RichardM678, 2017-07-24:
I am using unreal 4.15 and I do have Pluhins\Airsim folder under my unreal folder. I just right click and run it with launch game. I have Build.cs file in the Airsim, but it just doesn't work.

sytelus, 2017-07-27:
During last few days we been busy fixing EIGEN_ROOT and other setup issues that were reported. You might want to try the latest code. On Windows, its working great and very stable. On Ubuntu 16 as well its working but few known issues (see [docs](https://github.com/Microsoft/AirSim/blob/master/docs/linux_build.md)). Note that we have upgraded to unreal 4.16 so you will need that version. See our [install docs for Windows](https://github.com/Microsoft/AirSim/blob/master/docs/build.md).

closed 2017-07-27 by sytelus