microsoft/AirSim #362: camera pose clarifiction

ShukuiZhang, 2017-07-21:
Hi, I'm currently using the scene and depth image and camera poses to generate matching features. I found the camera pose is not well clarified. I think when I call getPosition(), the position returned is the drone position represented in global (NED) frame, however, when I call getOrientation(), the quaternion returned is the rotation represented in the drone frame (I had a test and this make more sense rather than in global frame. Am I right?). 

The drone frame is a left handed one, since all other frames are in right handed frames, I would like to make everything in right-handed ones for pose computation. Now the question is: should I convert the returned quaternion to be the one in right hand frame (i.e. flipping the x,y axis according to [answers](https://stackoverflow.com/questions/32438252/efficient-way-to-apply-mirror-effect-on-quaternion-rotation) )or it already is? In other words, what does the returned quaternion really represent? Thanks.

sytelus, 2017-07-22:
Both position and quaternion are in NED frame in SI units. As NED is left handed system, if you want to convert to right handed system then flip z in position and x,y for quaternion.

closed 2017-07-22 by sytelus

ShukuiZhang, 2017-07-22:
NED is left handed one?? 
And also,  the quaternion returned represents the rotation from global frame to the drone frame, right?