microsoft/AirSim #363: Feature matching between images does not work reliably

ShukuiZhang, 2017-07-22:
Hi all,
    I'm trying to implement AirSim to generate matched features between images, given the known ground truth poses and the camera model. However, the matching error between features shows up from time to time. The following is my matlab code to do the matching: 
```
close all
clear all
clc

%% Assumptions: 
% pose accuracy from file 
% synchronization
% intrinsic matrix and distortion???

corner_num = 20;
img_path = 'C:\\Users\\shzhang\Documents\AirSim\data_8\';

ind_1 = 1;
% load 1st scene image and depth image, depth is already converted to plane depth
% in c code 
if ind_1/10 <1
    img1 = imread([img_path, 'left_00000', num2str(ind_1), '.png']);
    img1 = rgb2gray(img1);

    [depth1, a_] = read_pfm([img_path, 'depth_00000', num2str(ind_1),'.pfm']); 
    depth1 = 255* depth1;
    depth1_crop = depth1(1: size(img1, 1), :); % make it same size as scene image
else
    img1 = imread([img_path, 'left_0000', num2str(ind_1), '.png']);
    img1 = rgb2gray(img1);

    [depth1, a_] = read_pfm([img_path, 'depth_0000', num2str(ind_1),'.pfm']);
    depth1 = 255* depth1;
    depth1_crop = depth1(1: size(img1, 1), :);
end
    
% data format each row: [timestamp position(3) quaternion(4)]
pose_data = csvread([img_path, 'data_file.csv']);

%% Here I denotes the drone frame (making it I for IMU, which is assumes to be coincide with the drone frame)
% c the camera frame, G the global (NED) frame
% camera position in drone frame
I_p_c = [0.46, -0.125, 0]';

% I made the drone frame a right handed one by flipping the z axis, the
% camera frame is conventional: with x pointing right, y pointing down and
% z pointing into image with origin to be camera center
I_c_R = [0 0 1;
         1 0 0;
         0 1 0]; 

% drone position in global frame
G_p_I1 = pose_data(ind_1, 2:4)';

% AirSim give I_G_q denoting the rotation from global frame to drone frame
img1_orientation = pose_data(ind_1, 5:8); %% quaternion format: [x y z w]
% making it to be from drone frame (I) to global frame (G)
img1_orientation = quat_inverse(img1_orientation);

G_I1_R = quat2rotmat(img1_orientation); % converting to rotation matrix
G_c1_R = G_I1_R* I_c_R; % rotation from camera frame to global frame
G_p_c1 = G_I1_R* I_p_c + G_p_I1; % position of camera origin in global frame

%% estimate the intrinsics, here I use the simplified model, which neglects
% the distortion 
height = size(img1, 1);
width = size(img1, 2);

u0 = width/2;
v0 = height/2;
f = width/2;

intrinsic_mat = [f 0 u0;
                 0 f v0;
                 0 0 1 ];

% detect corners in first scene image
corners = detectMinEigenFeatures(img1);
corners1 = corners.selectStrongest(corner_num);

figure(100)
imshow(img1)
hold on
plot(corners1) % plot the corners in the image as green cross

% This is to generate a few images to compare, here only use 1 image
for ind = 5: 5
    if ind_1 == ind
        continue
    end
    
    % index for the second image to compare
    ind_2 = ind;
    img2 = imread([img_path, 'left_00000', num2str(ind), '.png']);
    img2 = rgb2gray(img2);

    % camera pose for the second scene image, similar to the first one
    G_p_I2 = pose_data(ind_2, 2:4)';
    img2_orientation = pose_data(ind_2, 5:8);
    img2_orientation = quat_inverse(img2_orientation); 

    G_I2_R = quat2rotmat(img2_orientation); 
    G_p_c2 = G_I2_R* I_p_c + G_p_I2;
    G_c2_R = G_I2_R* I_c_R; 

    % relative camera pose betwenn scene1 and scene2
    c2_c1_R = G_c2_R'* G_c1_R; % relative rotation
    c2_p_c1 = G_c2_R'* (G_p_c1 - G_p_c2); % relative position

    pc_c2 = []; % point cloud
    feature2_location = [];
    pc_c1 = [];

    for i = 1: corners1.Count
        % return location is (x, y), i.e. [column, row] in image
        location = round(corners1(i).Location);
        if abs(depth1(location(2), location(1))- max(max(depth1))) < 0.1
            continue
        end
        %% project feature points in first scene image to 3D points
        c1_points_3D_unit = intrinsic_mat\[location(1); location(2); 1];
        c1_points_3D = depth1(location(2), location(1))* c1_points_3D_unit;
        pc_c1 = [pc_c1, c1_points_3D];

        c2_points_3D = c2_c1_R* c1_points_3D + c2_p_c1;
        uv2_tmp = intrinsic_mat* c2_points_3D;

        feature2_location = [feature2_location, uv2_tmp(1:2)/ uv2_tmp(3)];
        pc_c2 = [pc_c2, c2_points_3D];
    end

%     corners2 = detectMinEigenFeatures(img2);
%     corner2 = corners2.selectStrongest(corner_num);

    figure(ind)
    imshow(img2)
    hold on
    % plot the reprojected points in 2nd scene image
    plot([feature2_location(1,:)], [feature2_location(2, :)], 'ro');
%     plot(corner2);
    
end

function A = quat2rotmat(q) % quaternion to rotation matrix
    %% quaternion format: [x y z w]
    A(1, 1) = q(1) * q(1) - q(2) * q(2) - q(3) * q(3) + q(4) * q(4);
    A(1, 2) = 2 * (q(1) * q(2) + q(3) * q(4));
    A(1, 3) = 2 * (q(1) * q(3) - q(2) * q(4));

    A(2, 1) = 2 * (q(1) * q(2) - q(3) * q(4));
    A(2, 2) = -(q(1) * q(1)) + q(2) * q(2) - q(3) * q(3) + q(4) * q(4);
    A(2, 3) = 2 * (q(2) * q(3) + q(1) * q(4));

    A(3, 1) = 2 * (q(1) * q(3) + q(2) * q(4));
    A(3, 2) = 2 * (q(2) * q(3) - q(1) * q(4));
    A(3, 3) = -(q(1) * q(1)) - q(2) * q(2) + q(3) * q(3) + q(4) * q(4);
end

function q_inv = quat_inverse(q)
    assert(size(q, 2) == 4);
    q_inv = [- q(:, 1:3), q(:, 4)];
end
```

The extracted features in the first image look like this: 
[image]
While the projected features as red circles as follows:
[image]
As we can see, many of the features have a salient mismatch. I think my code is okay. So what could be the problem? My thinking is:
1. the camera poses may not be accurate enough 
2. scene and depth synchronization issue may still exist (Actually, I noticed this issue occasionally, even with the latest code, but in this test, I make the depth and scene image synchronized well)
3. Can I really use the simplified camera model for the matching or it is over-simplified? 
Thank you so much!

sytelus, 2017-07-22:
The depth image from Unreal is currently not planner depth. We have [some python code here](https://github.com/Microsoft/AirSim/blob/master/PythonClient/generate_stereo_data.py#L20) to convert it to planner depth but I've neither tested it or can attest if its right. So my question is how are you converting Unreal's depth to planner depth.

If you suspect pose is potentially wrong then I would suggest to insert sleep in your python code after calling simSetPose just to make sure Unreal has time to render. This shouldn't be needed but definitely one thing to try... If your scene is static then synchrnization between scene and depth image shouldn't matter.

ShukuiZhang, 2017-07-22:
I noticed there is the ConvertToPlanDepth function in your c code, I think that is reasonable, actually it functions the same as the python code you mentioned above.

simSetPose is only for computer vision mode right? Previously, I use the RC transmitter to control the drone and rarely have that involved. It might be an option, but I noticed big jump between poses, which make the trajectory not such realistic. But I will try to see if it helps. Thank you!

ShukuiZhang, 2017-07-25:
When you say "scene is static", do you mean when it is in computer vision mode? How about when using RC in HIL?

ShukuiZhang, 2017-07-26:
I tried in computer vision mode and uses the keyboard to move the drone. The depth looks reasonable, but the relative pose is still confusing. Have running out of idea what to do for the poses to make the features matched well ....... @sytelus

sytelus, 2017-07-27:
My guess is that it has something to do with how planner depth is computed. In about function, there is parameter f which I'm not sure how it was computed. Basically, we have perspective depth and FOV=90 and whatever resolution you are using. Using these 3 quantities you need to compute depth in camera plan.

Are you using stereo images or just mono? If you are using stereo images, then you will need camera distance which is 25 cm. For mono images, it doesn't matter.

ShukuiZhang, 2017-07-27:
Hi @sytelus , I'm using Mono. I tried to skip the depth map, which may be problematic and using the pose only, but the camera pose I computed never make sense to me.
The way I compute the relative camera poses are as follows, please correct me if I'm wrong:
1) grab the position and quaternion returned from the recording, I'm assuming they are drone poses represented in global (NED, which is right hand (RH) coordinates) frame. I assume the quaternion uses the Hamiltion convention.
2) The drone frame is a left handed one, so I tried to flip the z axis, which make it RH.  To do that, according to your earlier explanation, I need to flip the z in returned position and x, y in returned quaternion. Then now I have the drone pose (RH) in global frame.
3) Since the our convention for camera frame is x for right, y for down and z into image from camera center, I then did regular pose transformation to get the camera pose in global frame from the known relative pose of drone and camera. Given the above frame definition, camera position in drone (RH) frame is I_p_c = [0.46, -0.125, 0] and 
rotation from camera to drone is:
I_R_c = [0 0 1
             1 0 0
             0 1 0]
Is this correct? Thanks!
By the way, I doubt the parameter f value to be simply computed by width/2 ......

sytelus, 2017-07-27:
I'm pretty confused by your description and lot of things don't look right,

1. Camera is attached to the drone and both frames have just difference in x and y but otherwise everything else is same. Both frames are in NED.

2. In Computer Vision mode you can ignore drone altogether. The simGetImages API returns you ImageResponse that has camera position and orientation in NED. So no need to worry about drone frame and no need to do any additional transformation for drone frame.

3. In NED: front is +X, right is +Y and down is +Z. This is for camera as well as drone.

4. If your code assumes camera to be in different frame then you will need transformation.

ShukuiZhang, 2017-07-27:
Hi @sytelus  according to your description, every frame (global, drone, camera) is in NED, thus there is no need for the left right conversion anywhere???

sytelus, 2017-07-28:
Yes, that's correct. If your own code assumes NED then no transformations needed. BTW, I'd been talking with my colleague about planner conversion for depth and we have figured out that our current formulation is incorrect. We derived new formula for it but haven't tested it. So I would strongly suggest to think about that part.

ShukuiZhang, 2017-07-28:
Thanks for pointing that out. @sytelus That may account for the mismatch I raised in the beginning. Please let me know when you have the depth code ready. Thanks.

ShukuiZhang, 2017-07-28:
Hi @sytelus . I had a few test for the epipolar constraint, it skips the depth map. Basically, if camera poses and intrinsic parameters are accurate enough, the error for the test should be small. But I find that error level is not consistent, i.e. for half of the times, the errors are small, but for the rest, it is not. Due to the good half, I tend to believe the intrinsics are good. Then it means the generated poses are not accurate enough from time to time. Do you think that may happen? What might be the problem?

These are the few lines that is responsible for capturing, which is similar to the original code:
```
while(sample < num_samples) {
                //const auto& collision_info = client.getCollisionInfo();
                //if (collision_info.has_collided) {
                //    pose_generator.next();
                //    client.simSetPose(pose_generator.position, pose_generator.orientation);

                //    std::cout << "Collison at " << VectorMath::toString(collision_info.position)
                //        << "Moving to next pose: "  << VectorMath::toString(pose_generator.position)
                //        << std::endl;

                //    continue;
                //}
                ++sample;
				std::cout << sample << std::endl;
                auto start_nanos = clock->nowNanos();

                std::vector<ImageRequest> request = { 
                   // ImageRequest(0, ImageType_::Scene, false),
                    ImageRequest(1, ImageType_::Scene, false),
                    ImageRequest(1, ImageType_::Depth, true) //true
                };
                const std::vector<ImageResponse>& response = client.simGetImages(request);
                if (response.size() != 2) {
                    std::cout << "Images were not recieved!" << std::endl;
                    start_nanos = clock->nowNanos();
                    continue;
                }

                ImagesResult result;
                result.file_list = &file_list;
				result.filestream = &filestream;
				result.filestream1 = &filestream1;
                result.response = response;
                result.sample = sample;
                result.render_time = clock->elapsedSince(start_nanos);;
                result.storage_dir_ = storage_dir_;
				//result.position = response[0].camera_position;
				//result.orientation = response[0].camera_orientation;
				//result.position = pose_generator.position;
				//result.orientation = pose_generator.orientation;

                results.push(result);
				clock->sleep_for(1);
                //pose_generator.next();
                //client.simSetPose(pose_generator.position, pose_generator.orientation);
            }
```
I didn't use simSetPose because that makes the trajectory weird. BTW, the pose is obtained through ImageResponse returned camera position and orientation. 

One last issue is that when I request the scene and depth image at the same time, I will get two camera poses corresponding to each response, which one should I use? Sometime, I find the difference is not trivial. Thanks.

sytelus, 2017-07-29:
Could you insert clock->sleep_for(1) before the line `ImagesResult result;`? That should eliminate pose difference. Also eliminate while loop if you are going to just take one image. Basically try to simplify as much possible.

sytelus, 2017-11-14:
The depth generation had been revamped in new releases with both planner and perspective depth available now: https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md#what-does-pixel-values-mean-in-different-image-types

If you still see problem, please file new issue with info on how to reproduce the problem.

closed 2017-11-14 by sytelus

sytelus, 2018-02-21:
FYI – I just added ability to get camera parameters via APIs in AirSim (1st bullet point below). Here’s [more info in our docs](https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md#getting-camera-parameters).

diarmaidocualain, 2018-11-13:
Hi @ShukuiZhang, I am wondering if you ever resolved your issue on this? If so, how? 
I am having a similar issue mapping from my Airsim camera location and pose to my Matlab environment and I am wondering if I have the same problem. You can see where the corners of the TV in the Unreal "Realistic Rendering" environment dont line up with the points in the perspective image highlighted below. 
[image]

Just to add to some of your questions: 

> 1. grab the position and quaternion returned from the recording, I'm assuming they are drone poses represented in global (NED, which is right hand (RH) coordinates) frame. I assume the quaternion uses the Hamiltion convention.

Everything returned by the Airsim simulator is in NED coordinates. This includes the camera coordinates/pose, drone coordinates/pose and so on. Airsim returns the quaternion as individual elements of a struct labelled "x", "y", "z" and "w", so neither Hamilton or any other convention is used as it is not needed. 

> 2. The drone frame is a left handed one, so I tried to flip the z axis, which make it RH. To do that, according to your earlier explanation, I need to flip the z in returned position and x, y in returned quaternion. Then now I have the drone pose (RH) in global frame.

No, all frames are right handed as they use NED. The Unreal Environment used left hand, but as Airsim only returns NED coordinates, you should not encounter this. Airsim does the conversion for you. As such, be wary of flipping only one axis such as z as it will change the handeness of the coordinates. If you flip two such as z and y together the handiness should remain the same. I am not sure myself what the mathematical rules are for this are for quaternions.