microsoft/AirSim #364: Setting GPS origin

sifengyang, 2017-07-24:
I want to run  takeoff.py .  so I have to install TISL by sitl.md. I have completed the step ,"Software-In-Loop Simulation (SITL)". When I faced “Setting GPS origin”，I do not know how i do it.
[image]

fountainYuan, 2017-07-24:
Input these command in SITL console

sifengyang, 2017-07-24:
@fountainYuan I use ubuntu system . how to I can find SITL console. thanks

fountainYuan, 2017-07-24:
./build_posix_sitl_default/src/firmware/posix/px4 ./posix-configs/SITL/init/ekf2/iris

after connect AirSim 

press enter 

you will see

then input param set command

sifengyang, 2017-07-24:
I can't do it by your way. I can't input the command under normal circumstances. Am I doing it wrong? however I find another problem on unreal. the message is "not receiving any message from HIL, please restart your HIL node and try again ".
[image]

lukaszdworako, 2017-07-24:
That message only occurs when you are not using a pixhawk hardware controller, this is not an issue and should be nothing to worry about. What do you mean by "can't input under normal circumstances"?

sifengyang, 2017-07-25:
@lukaszdworako  I am so sorry . it mean  is that I press  enter, then  input param set command, but SITL console prints 
"WARN  [lib__ecl] EKF baro hgt timeout - reset to baro
WARN  [lib__ecl] EKF baro hgt timeout - reset to baro
INFO  [commander] Landing detected
INFO  [commander] Takeoff detected
INFO  [commander] Freefall detected
INFO  [commander] Landing detected
INFO  [commander] Takeoff detected
INFO  [commander] Freefall detected
" Constantly. 
[image]

fountainYuan, 2017-07-25:
you can pause the Unreal block project  then output will pause,you can input command .
Or you can copy command to console then press enter to excute command

closed 2017-07-25 by sifengyang