microsoft/AirSim #367: Unable to control the Drone on Mac OS

AndrewEj, 2017-07-25:
My setup :
System: MacOS
i5 CPU @2.00 GHz
8GB RAM
SSD harddrive
RC : No RC
Flight controller : SITL
UE 4 Version: 4.15.3

I update Airsim and PX4 to the latest version and got everything set up. But I cannot control the drone in UE4. I don't think there is any issues with PX4, because it can communicate with QGroundControl. 

This is what I got from UE4. 

[image]

This is the result from PX4 when I typed "commander status"
[image]

I cannot find the reason..
Please help .

AndrewEj, 2017-07-25:
This is what I got from  DroneShell. 

[image]

AndrewEj, 2017-07-25:
I try to setup Airsim on windows7. I used PX4 SITL on Mac to remotely control the drone on win7. Everything seems to be OK. But it repeats hit the ground

sytelus, 2017-08-14:
First thing to try is make sure your PX4 setup is working with QGroundControl. That is, you should be able to arm, take off etc from QGC. If you can't then you have problem with PX4 setup and you probably want to report this to PX4 forums or Github issues at https://github.com/PX4/Firmware/issues.

closed 2017-08-14 by sytelus