microsoft/AirSim #369: AirSim Drone is unstable

SpeedCooper, 2017-07-25:
Hi. I am a student who is using the AirSIm program in Korea. During the use of the AirSim program, the drones are moving at their own discretion. Also use the 'Droneshelle.exe' program to type commands and move them around.
[image]

fountainYuan, 2017-07-25:
you must use PX4 Firmware 1.5.5, Higher is not suit for AirSim

SpeedCooper, 2017-07-25:
I used nuttx-px4fmu-v2-default.px4 of px4 V.1.4.4. Is not there a problem?

SpeedCooper, 2017-07-26:
As a result of running px4 version 1.5.5, the drones are strangely working alone.
[image]

fountainYuan, 2017-07-26:
According your picture , the GPS of your UAV is lost ,you simulation in SITL　or HITL ?   If is SITL, you can set GPS params with param set command  like "  param set LPE_LAT 47.641468 " "param set LPE_LON -122.140165"  the coord is Microsoft.
You maybe change number. Or you can pause the Unreal Engine for UAV get GPS coord then click Play to Sim.

SpeedCooper, 2017-07-26:
I'm using HITL.

SpeedCooper, 2017-07-26:
I'm using HITL.

SpeedCooper, 2017-07-26:
And does it matter if the wings are not fixed on the shaft?
[image]

fountainYuan, 2017-07-26:
I found newest AirSim maybe have bugs ,you can try
 git clone --branch VehicleCamera https://github.com/Microsoft/AirSim.git 
this banch can control in QGC.

SpeedCooper, 2017-07-28:
@fountainYuan 
I tried it, but it's not working. I don't know why....

fountainYuan, 2017-07-28:
you get any error message? I use PX4 Version 1.5.5 AirSim  banch Vehicle Camera , can use QGC  control flying.

sytelus, 2017-07-28:
Hey folks, I can't reproduce this but I have two guesses:

1. Your "Player Start" is too high above ground. This might cause drone to fall off and its initialization/calibration get messed up. So please make sure "Player Start" is just above the ground.

2. Another reason might be removal of ground lock which we did few weeks back. I've just added it back so try out new code as well.

SpeedCooper, 2017-07-28:
@sytelus 

What new code does it say? PX4? AirSim?

sytelus, 2017-07-29:
AirSIm. Please follow the build instructions: https://github.com/Microsoft/AirSim/blob/master/docs/build.md

closed 2017-07-29 by sytelus

SpeedCooper, 2017-07-31:
@sytelus 
I installed it as shown on the homepage, but it is the same.....

machenxiang, 2019-03-16:
I have the same question in HITL. My version of airsim is 1.2.1. I have tried the PX4 firmware 1.4.4,1.5.5and 1.8.2 but the question still exists.