microsoft/AirSim #370: Takeoff, Fall, Unable to control Drone

AndrewEj, 2017-07-25:
I try to use px4 on mac to control the drone on win7 remotely. But the following problems repeat. 

[image]

I tried to switch the version of PX4 and Airsim. But I don't know why Mac cannot make px4 1.5.5, and win7 cannot build earlier Airsim version.

AndrewEj, 2017-07-25:
When I use WIN7 to build this version : 6458eaf44aff5a802d9953457e06b0734f16d6b9 airsim; I can see this error : 
[image]

AndrewEj, 2017-07-25:
When I use Mac to build PX1.5.5, I can see this error : 
[image]

AndrewEj, 2017-07-27:
It seems the windows7 can compile the latest version, but not the previous one. Every time when I hit the play button, it immediately crashes. However, when I changed the value of the LocalIP, which I deliberately mis-connect the port, I can see the drone. I guess there are issues with the vehicle controller.

sytelus, 2017-07-27:
Could you please make sure your build config in Visual Studio is set to "DebugGameEditor" and "Win64". Some time its set to "Development" and it will crash.

closed 2017-07-27 by sytelus