microsoft/AirSim #372: The problem to control the drone flight

sifengyang, 2017-07-25:
Hi, guys. I have install PX4 and SITL. And I can control the drone flight by takeoff.py script. However, How to I control manually the drone flight without flight controller harware.

sytelus, 2017-07-25:
If you don't want to use flight controller, take a look at [Computer Vision mode](https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md). We are soon also going to enable simple_flight and ROSFlight that runs inside the AirSim.

closed 2017-07-25 by sytelus

sifengyang, 2017-07-26:
@sytelus thanks. I can control the drone flight . But I think that the drone flight normally. Propellers of the drone are  inactive.  
[image]