microsoft/AirSim #376: Quantization in the processing of depth images from the simulator?

Mayankm96, 2017-07-26:
Hey! 

So I have been trying to write my own ros wrapper to communicate with the AirSim (a bit of help from [here](https://github.com/marcelinomalmeidan/publishAirsimImgs). While I am processing the depth image using opencv and viewing the output using image_view, I am seeing quantization occurring over the image. Is it due to conversion of float to uchar in the `image_view`?

A snippet of the code for processing the depth (similar to the function [convertToPlanDepth](https://github.com/Microsoft/AirSim/blob/master/Examples/StereoImageGenerator.hpp)):
```
cv::Mat convertToPlaneDepth(const cv::Mat& input, float scale, float f = 320){
    int width = input.cols;
    int height = input.rows;

    float center_i = width / 2.0f - 1;
    float center_j = height / 2.0f - 1;

    cv::Mat output = cv::Mat(height, width, CV_32FC1);
    cv::Mat depth_float;
    input.convertTo(depth_float, CV_32FC1, scale);

    for (int i = 0; i < width; ++i) {
        for (int j = 0; j < height; ++j) {
            float dist = std::sqrt((i - center_i)*(i - center_i) + (j - center_j)*(j - center_j));
            float denom = (dist / f);
            denom *= denom;
            denom = std::sqrt(1 + denom);
            output.at<float>(j, i) = input.at<float>(j, i) / denom;
        }
    }
    return output;
}
```
where `input = cv::imdecode(response.at(2).image_data, CV_LOAD_IMAGE_GRAYSCALE)`.

[image].

Also for creating point clouds, do the camera paramters change if you are taking a 720p form the depth, and rgb rendered targets?

Thanks for your help!

sytelus, 2017-07-27:
Unreal generates the ground truth *perspective* depth image. The value of each pixel is a depth value from 0 to 2^24-1. However as this range is too large, the depth map would end up looking like all white. So we clip the depth at 10,000 cm and then normalize it from 0.0 to 1.0. 

I think what you need is depth from the plan. To get the planner depth you need to do additional processing. One attempt to do this  can be [found here](https://github.com/Microsoft/AirSim/blob/master/Examples/StereoImageGenerator.hpp#L200)). However I am not sure if its right. Especially I don't know how parameter f was calculated. If you know correct way to convert to planner depth given FOV and resolution of image, let us know.

Mayankm96, 2017-07-29:
Well the image I shared above is the one from the `convertToPlanDepth()` function. I published both the raw (from the airsim) and the one after this processing to see the difference between the two. I am now confused on the way the processed one always look bright at the center and becomes darker as we proceed towards the boundary while the one from the simulator looks "more accurate". 

At the same time, I might be confusing the meaning of a depth image with the real depth/distance of the objects from the camera.

The unprocessed depth image (from simulator):
[image]

The processed depth image (after processing from the `convertToPlanDepth()`):
[image]

sytelus, 2017-07-31:
I was going over convertToPlanDepth() implementation with one of my colleague and I think current one is wrong. We have derived new equation that looks more right and I'll probably re-check that and implement it around end of week. Meanwhile, if you are familiar with computer vision algos for how to convert perspective projection to planner projection given FOV and resolution, let us know your suggestions as well!

Mayankm96, 2017-10-05:
@sytelus I guess this issue is resolved now due to recent changes in the code. I'll be closing it now.

closed 2017-10-05 by Mayankm96