microsoft/AirSim #380: when I control the drone flight by python script. the Unreal app crash suddenly

sifengyang, 2017-07-27:
my goal is that I use the `moveToPosition` function  to control the drone can flight . the python script is that 
````
from PythonClient import *

import cv2

import time

import sys

import array

client = AirSimClient('127.0.0.1')

position_file = open("/opt/AirSim/PythonClient/drone_position.txt",'r')

#takeoff = client.takeoff(5)

#time.sleep(6)

i=0

for line in position_file.readlines():

	i = i+1

	position = line 

	#position = array(position)

	print float(position.split(",")[0].split("[")[1])

	client.moveToPosition(float(position.split(",")[0].split("[")[1]),float(position.split(",")[1])+10,float(position.split(",")[2].split("]")[0]),10, 0, DrivetrainType.ForwardOnly, YawMode(False,0), 20, 1)

	time.sleep(10)

	print i
position_file.close()
````
and the drone_position.txt is the following.
```
[-22.90005111694336, 0.5000002384185791, -0.14649581909179688]
[-22.90005111694336, 0.5000002384185791, -0.14649581909179688]
[-22.90005111694336, 0.5000002384185791, -0.14649581909179688]
[-22.90005111694336, 0.5000002384185791, -0.14649581909179688]
[-22.90005111694336, 0.5000002384185791, -0.14649581909179688]
[-26.800065994262695, 0.5000002384185791, -0.14649581909179688]
[-27.00006675720215, 4.7999982833862305, -0.14649581909179688]
[-27.00006675720215, 6.699996471405029, -0.14649581909179688]
[-26.000062942504883, 2.500000476837158, -0.14649581909179688]
[-21.800046920776367, 2.3000004291534424, -0.14649581909179688]
[-18.100032806396484, 2.8999998569488525, -0.14649581909179688]
[-18.100032806396484, 7.099996089935303, -0.14649581909179688]
[-14.000019073486328, 11.300004959106445, -0.14649581909179688]
[-12.20001220703125, 15.500020027160645, -0.14649581909179688]
[-12.20001220703125, 19.900035858154297, -0.14649581909179688]
[-12.20001220703125, 24.20005226135254, -0.14649581909179688]
[-12.20001220703125, 28.600069046020508, -0.14649581909179688]
[-12.20001220703125, 33.00006866455078, -0.14649581909179688]
[-12.20001220703125, 37.5000114440918, -0.14649581909179688]
[-12.20001220703125, 41.89997100830078, -0.14649581909179688]
[-12.20001220703125, 46.39990234375, -0.14649581909179688]
[-12.203502655029297, 50.89982604980469, -0.14649581909179688]
[-12.206993103027344, 55.59973907470703, -0.14649581909179688]
[-12.21048355102539, 60.299644470214844, -0.14649581909179688]
[-12.215719223022461, 65.19953155517578, -0.14649581909179688]
[-12.222700119018555, 70.09939575195312, -0.14649581909179688]
[-12.222700119018555, 74.99932098388672, -0.14649581909179688]
[-12.222700119018555, 79.89924621582031, -0.14649581909179688]
[-12.222700119018555, 84.7991714477539, -0.14649581909179688]
[-12.219209671020508, 89.69906616210938, -0.14649581909179688]
[-12.217464447021484, 94.5989761352539, -0.14649581909179688]
[-12.217464447021484, 99.69889831542969, -0.14649581909179688]
[-12.21048355102539, 104.69876098632812, -0.14649581909179688]
[-12.206993103027344, 109.79865264892578, -0.14649581909179688]
[-12.206993103027344, 111.49862670898438, -0.14649581909179688]
```
my problem is that  when I tun the script ,the drone can fly for a distance,then the unreal app crash. I don't  know the reason.

sytelus, 2017-07-27:
What OS are you using? Are you using latest code?

sifengyang, 2017-07-28:
I use the ubuntu 14.04 system . but I don't understand meaning of the question  "Are you using latest code".

lukaszdworako, 2017-07-28:
Are you using the latest build of AirSim posted on github? @sifengyang

sytelus, 2017-07-28:
I would suggest to use Ubuntu 16.04. Old code crahses due to bug in Unreal. In latest code we have workaround. See our new Linux build instructions as well.

closed 2017-07-28 by sytelus

sifengyang, 2017-07-30:
@sytelus @lukaszdworako  I think my code might be wrong .

sifengyang, 2017-07-30:
@lukaszdworako  yes . I use the latest build of Airsim posted on github.