microsoft/AirSim #381: New Linux build

viniciusguigo, 2017-07-27:
Hello, guys.

First of all, thank you very much for releasing AirSim to the public and working on the Linux version of it! It is an amazing piece of software! Good job all devs! :)

One minor thing to add: could you please add a "sudo apt install cmake" on the beginning of the setup or build script? That's the only thing missing so it can install all dependencies.

I will be running extensive tests during this weekend and will report anything else I can contribute.

Thanks one more time!

sytelus, 2017-07-28:
Welcome and thanks for the suggestion. I'll put this in. The reason we hadn't because Ubuntu default install should already have it. In any case, would look forward for more feedback and suggestions!

closed 2017-07-28 by sytelus

Mayankm96, 2017-07-30:
I am trying to connect to PX4 SITL using the new AirSim built on branch `4.16`. While I follow the guidelines [here](https://github.com/Microsoft/AirSim/blob/master/docs/sitl.md), my PX4_SITL is still waiting for initial data on UDP port 14560. 
```
commands file: ./posix-configs/SITL/init/ekf2/iris
creating new parameters file
creating new dataman file

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

18446744073709551615 WARNING: setRealtimeSched failed (not run as root?)
WARN  [param] error importing from 'rootfs/eeprom/parameters'
Command 'param' failed, returned 1
  SYS_AUTOSTART: curr: 0 -> new: 4010
INFO  [dataman] Unkown restart, data manager file 'rootfs/fs/microsd/dataman' size is 47640 bytes
  BAT_N_CELLS: curr: 0 -> new: 3
  CAL_GYRO0_ID: curr: 0 -> new: 2293768
  CAL_ACC0_ID: curr: 0 -> new: 1376264
  CAL_ACC1_ID: curr: 0 -> new: 1310728
  CAL_MAG0_ID: curr: 0 -> new: 196616
  CAL_GYRO0_XOFF: curr: 0.0000 -> new: 0.0100
  CAL_ACC0_XOFF: curr: 0.0000 -> new: 0.0100
  CAL_ACC0_YOFF: curr: 0.0000 -> new: -0.0100
  CAL_ACC0_ZOFF: curr: 0.0000 -> new: 0.0100
  CAL_ACC0_XSCALE: curr: 1.0000 -> new: 1.0100
  CAL_ACC0_YSCALE: curr: 1.0000 -> new: 1.0100
  CAL_ACC0_ZSCALE: curr: 1.0000 -> new: 1.0100
  CAL_ACC1_XOFF: curr: 0.0000 -> new: 0.0100
  CAL_MAG0_XOFF: curr: 0.0000 -> new: 0.0100
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  COM_RC_IN_MODE: curr: 0 -> new: 1
  NAV_DLL_ACT: curr: 0 -> new: 2
  COM_DISARM_LAND: curr: 0 -> new: 3
  NAV_ACC_RAD: curr: 10.0000 -> new: 2.0000
  COM_OF_LOSS_T: curr: 0.0000 -> new: 5.0000
  COM_OBL_ACT: curr: 0 -> new: 2
  RTL_RETURN_ALT: curr: 60.0000 -> new: 30.0000
  RTL_DESCEND_ALT: curr: 30.0000 -> new: 5.0000
  RTL_LAND_DELAY: curr: -1.0000 -> new: 5.0000
  MC_ROLLRATE_P: curr: 0.1500 -> new: 0.2000
  MC_PITCHRATE_P: curr: 0.1500 -> new: 0.2000
  MC_PITCH_P: curr: 6.5000 -> new: 6.0000
  MC_ROLL_P: curr: 6.5000 -> new: 6.0000
  MPC_HOLD_MAX_Z: curr: 0.6000 -> new: 2.0000
  MPC_Z_VEL_P: curr: 0.2000 -> new: 0.6000
  MPC_Z_VEL_I: curr: 0.0200 -> new: 0.1500
  EKF2_GBIAS_INIT: curr: 0.1000 -> new: 0.0100
  EKF2_ANGERR_INIT: curr: 0.1000 -> new: 0.0100
  EKF2_MAG_TYPE: curr: 0 -> new: 1
INFO  [platforms__posix__drivers__ledsim] LED::init
INFO  [platforms__posix__drivers__ledsim] LED::init
INFO  [simulator] Waiting for initial data on UDP port 14560. Please start the flight simulator to proceed..
```
Any idea on what might be wrong?