microsoft/AirSim #384: ERROR When Building AirSim on Ubuntu 16.04

lqsheng, 2017-07-28:
I am following the [tutorial](https://github.com/Microsoft/AirSim/blob/master/docs/linux_build.md) to build Airsim on Ubuntu 16.04.Everything seems OK.But when I go to Unreal folder and bring up Unreal Editor,it says that there isn't corresponding folder exists.
[image]
[image]
In the first three steps, I didn't receive any errors.Everything seems OK.Does anyone know how to solve this problem?

sytelus, 2017-07-28:
Looks like ou cloned AirSim inside Unreal. You want to move AirSim outside of Unreal folder. Then bring up Unreal Editor, click Browse, navigate to AirSim/Unreal/Environments/Blocks folder and then clock on Blocks.uproject.

closed 2017-07-28 by sytelus

viniciusguigo, 2017-07-28:
The UE4Editor is inside the Unreal Engine folder, which is different from the AirSim folder. I'd clone AirSim to a folder outside Unreal Engine.

lqsheng, 2017-07-28:
I had try to delete AirSim in UnrealEngine folder and then git clone AirSim in the GitHubSrc folder.The path shows in the picture below.
[image]
However,when I run `./build.sh`,I got the errors shows in the pictures below
[image]
The error is `error: 
     cannot pass object of non-trivial type 'std::string' (aka
      'basic_string<char, char_traits<char>, allocator<char> >') through
      variadic function; call will abort at runtime [-Wnon-pod-varargs]
  ...port %d, local IP %s, remote IP...", port, connection_info_.local_host_i...`

Besides,I can't find the correct place to run `UE4Editor`.I can't find `Engine/Binaries/Linux`folder in `/GitHubSrc/AirSim/Unreal`,But I found `/GitHubSrc/UnrealEngine/Engine/Binaries/Linux`,However,it still reminds me that can't find `UE4Editor` command.What goes  wrong? @sytelus

lqsheng, 2017-07-28:
I had try to delete AirSim in UnrealEngine folder and then git clone AirSim in the GitHubSrc folder.The path shows in the picture below.
[image]
However,when I run `./build.sh`,I got the errors shows in the pictures below
The error is `error: 
     cannot pass object of non-trivial type 'std::string' (aka
      'basic_string<char, char_traits<char>, allocator<char> >') through
      variadic function; call will abort at runtime [-Wnon-pod-varargs]
  ...port %d, local IP %s, remote IP...", port, connection_info_.local_host_i...`

Besides,I can't find the correct place to run `UE4Editor`.I can't find `Engine/Binaries/Linux`folder in `/GitHubSrc/AirSim/Unreal`,But I found `/GitHubSrc/UnrealEngine/Engine/Binaries/Linux`,However,it still reminds me that can't find `UE4Editor` command.What goes  wrong? @sytelus

Mayankm96, 2017-07-30:
I faced a similar issue right now. To fix it I switched to an alternate branch 4.16 on which my plugin got built just fine. Moreover, I am using Ubuntu 14.04. 

Just simply do: `git clone -b 4.16 https://github.com/Microsoft/AirSim.git`, and rebuild again. I hope this works for you as well.