microsoft/AirSim #387: How to fix Image aliasing issue?

ShukuiZhang, 2017-07-29:
Hi,
    I've been recording images for the dataset. However, it looks the images I recorded has significant aliasing. E.g. I get the image below:

[image]
and zooming the image, you can see the aliasing as below:
[image]

This prevents us to go any further for image features. Is there any way that I can maximize the image quality? I tried "pixel_as_float" and uncompressed version, but neither seems to work well... Thanks.

Robin-Wong, 2017-07-30:
Try to modify the settings.json for large resolution and please remember the rpcClient get RGBA image, unless you modify the file RenderRequest.cpp for uncompress. For pixel_as_float, I think AirSim only trans R channel with float32.

ShukuiZhang, 2017-07-30:
Thanks for your reply. @Robin-Wong I checked the code and you are right. But did you meet the same problem and fix them at your end? I don't think changing the resolution can make a change (Actually, I did so but no help). 

On the other hand, it looks that this may be a problem more about AirSim because if I take direct screenshot of the scene in the main window, I can see much less jaggies...

sytelus, 2017-07-31:
This very much looks like lack of resolution (as you are zooming). Also see here: https://answers.unrealengine.com/questions/293017/scenecapturecube-and-its-anti-aliasing.html. Note that images from Unreal takes much more longer to generate as you increase resolution.

ShukuiZhang, 2017-07-31:
Hi @sytelus The link you provided seems promising. Can you give some hint where to find the "scene capture 2D actor"? really new to UE4 and did not find the result after some searching online. Thanks.

sytelus, 2017-07-31:
Render target gets set up here: https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/PIPCamera.cpp#L122

Same function also sets up SceneCapture2D (which is just like camera but allows to get image out of it).

You can set properties there as you like and experiment. I was searching about SceneCapture2D and anti-aliasing and there seems to be lots of forum posts that anti-aliasing wasn't working. I would suggest to take look at them as well just to see if any of the settings are missing.

ShukuiZhang, 2017-07-31:
I have been searching for the anti-aliasing stuff for yesterday, even with the best settings I can have, the captured image still has jaggy. I also tried to modify the settings @sytelus mentioned, with width = 1920 and height = 1080, I can see the improvement, but seems still not good enough. The following is the captured image:
Default PIPCamera.cpp:
[image]

After tuning settings per @sytelus :
[image]

The raw screenshot from UE4:
[image]

Out of the three, it looks the ue4 screenshot has the best quality of all.

adamkeynes, 2018-05-03:
meeting same problem. i changed AA method in my ue4 project but sub window images never change.  
there should have some params about setting anti-aliasing in sub window rendering.  maybe we just haven't locate it, or it's not configured so a default params with no AA method is activated

jonyMarino, 2020-09-04:
Closed due to age and incompleteness.
Please feel free to open a new issue (refering this one) if you still have this problem, and include the following (as requested when opening a bug issue)

*Make sure to write all reproduction steps
*Include full error message in text form
*Search issues for error message before filing issue
*Attach screenshot if applicable
*Include code to run if applicable

Thanks!

paulerikf, 2021-10-01:
@jonyMarino I realize this issue is closed, but hopefully it's ok if I still post this here:

Adding `capture->ShowFlags.SetTemporalAA(true);` before line [402](https://github.com/microsoft/AirSim/blob/2b01754aad58cafa5b68308a9c72482ca6baf8b9/Unreal/Plugins/AirSim/Source/PIPCamera.cpp#L402) got AntiAliasing working for me.

paulerikf, 2021-11-01:
@jonyMarino I've been away for a bit, sorry. I hesitated to place a PR because I wasn't sure if having AA should be the default behavior, or if it should be configurable in some way... (if so, I haven't put in the time to figure out how that'd be done).

If having AA hardcoded on is fine I can easily add a PR, if it should be configurable I'd need some time to figure that out.

jonyMarino, 2021-11-12:
@paulerikf You are right. This should be configurable. This impacts performance, and I don't know if it affects another type of image like the depth one.
The settings.json would be the right place to create a new parameter for this.