microsoft/AirSim #391: Built in flight control

ZixingLin, 2017-08-01:
Hi, I have installed AirSim on unreal engine and the drone has spawned when I click play as it is supposed to. However there is a problem that I am facing, I am running on a Windows 7 computer and have no hardware controllers as required. Pardon me for my silly question but is it possible to control the drone directly without setting up other things - eg PX4 SITL. I read something about being able to fly the drone with a ROSFlight controller that exist in the simulation but am unable to find much information about it. 

Thank you for attending to this question and I look forward any replies.

fountainYuan, 2017-08-01:
Do you mean fight control with program in VS ?

ZixingLin, 2017-08-01:
Sorry if I was being vague, basically what I meant is that I would like to know if there are ways to pilot the drone in the simulation without using SITL/PixHawk hardware. The main reason I'm asking this is because I'm running on Windows 7, and bash isn't available on this version of windows which makes it seem impossible to get the simulation working at all since I'm unable to pilot my drone in any way without it, or so it seems.

fountainYuan, 2017-08-01:
You can look at https://px4.osdrone.net/1_Getting-Started/windows.html

ZixingLin, 2017-08-08:
Thank you for your reply fountainYuan. However in order to get SITL to work I either need a Linux OS, or a Windows 10 computer with bash installed - the issue is that I'm on a Windows 7 computer hence that would not be possible for me.

I was reading about the SimpleFlight mechanic getting developed to remove the need of PX4, is that out yet?

closed 2017-08-11 by ZixingLin