microsoft/AirSim #392: Programmatically send backspace from Python script to reset/respawn drone position?

gmgremillion, 2017-08-01:
@sytelus You mentioned (in [#331](https://github.com/Microsoft/AirSim/issues/331#issuecomment-314369112)) using backspace to respawn the vehicle back at the player start location/origin, which I'm assuming is the method demonstrated in @lovettchris' [video](https://youtu.be/d_FyjKDWQfc?t=1m50s) (at 2:01). I am able to replicate this myself with a SITL setup. 

How might you send this backspace command programmatically from a python script/API, given that it specifically requires clicking on the Unreal simulation window? Is there an existing way to toggle the depth/segmentation/scene mini-windows from a script, which I could exploit for this purpose? I'm using SITL to avoid having to physically restart the pixhawk, possibly using a mavlink command to reset SITL (as suggested in [#276](https://github.com/Microsoft/AirSim/issues/276#issuecomment-308551443)). 

@viniciusguigo

sytelus, 2017-08-03:
This is a great suggestion. I think we can put an API for this. BTW, I'm almost about to complete simple_flight firmware which is integrated with AirSim and eliminates the need for PX4. This new API would work great with it as well.

KeeganVG, 2017-08-11:
Has there been any developments with this? I would be very interested if there was.

sytelus, 2017-11-14:
reset() API is available now: https://github.com/Microsoft/AirSim/blob/master/docs/apis.md#common-apis

closed 2017-11-14 by sytelus