microsoft/AirSim #393: PIXHAWK HIL erratic behavior during offboard control

saihv, 2017-08-01:
This issue is somewhat along the lines of #316 and #291, and mainly deals with problems during offboard control. I am using the latest version of AirSim, PIXHAWK in hardware in the loop and PX4 firmware 1.5.5 as suggested in #291. Parameters were reset before the experiment and the RC was recalibrated. In order to make sure the PX4 does not enter failsafe, I set NAV_RCL_ACT and NAV_DLL_ACT to 'disabled'. 

Mainly, the problem is that the PIXHAWK does not respond properly to offboard commands. After arming, at some point, the main LED starts flashing purple rapidly (which usually means failsafe), and at the same time, the B/E led on the PIXHAWK (FMU) also starts flashing amber rapidly. This happens during both manual flight and offboard mode, but it doesn't seem to affect manual flight as much as it does offboard. In offboard mode, takeoff commands are not dealt with properly: the drone shoots up and sometimes comes back down again to land, and position commands don't work properly as well.

I am attaching some logs (both MavlinkTest logs as well as PX4 native logs) that were captured during my tests with both manual and offboard mode control. I had to use the older sdlog2 way of logging because SYS_LOGGER=1 as suggested in logging.md resulted in no logs for me. 

In flight 1, I only armed the PIXHAWK through the RC and no further commands were given. Within seconds of arming, the PIXHAWK was flashing purple and the B/E LED was flashing orange.

[MavlinkTest_RC_port14389.zip](https://github.com/Microsoft/AirSim/files/1191912/MavlinkTest_14389.zip)
[MavlinkTest_RC_port14550.zip](https://github.com/Microsoft/AirSim/files/1191913/MavlinkTest_14550.zip)
[PX4 log RC flight](https://github.com/Microsoft/AirSim/files/1191946/PX4_RC.zip)
[PX4 log file visualized](http://logs.uaventure.com/view/qYt5EjYgKtAVjuFqXuJZx8)

In flight 2, I armed the PIXHAWK through DroneShell, and then issued a takeoff command. 

[MavlinkTest_Offboard_port14389.zip](https://github.com/Microsoft/AirSim/files/1191936/MavlinkTest_O_14389.zip)
[MavlinkTest_Offboard_port14550.zip](https://github.com/Microsoft/AirSim/files/1191937/MavlinkTest_O_14550.zip)
[PX4 log offboard flight.zip](https://github.com/Microsoft/AirSim/files/1191947/PX4_OB.zip)
[PX4 log file visualized](http://logs.uaventure.com/view/NAZUJCqL6tX4eaYQDdPwei)

Note, in case it's useful information: I am using an older version of AirSim for my multi-vehicle fork AirSimMulti, which uses PX4 firmware ~1.4.4. In these version, with the same PIXHAWK board, manual flight works well (no problem with the B/E led etc.), and I can fly using auto mode as well through the RC. But I have similar problems with offboard mode, and with the PIXHAWK LED flashing purple. 

Looking forward to your thoughts about this!

viniciusguigo, 2017-08-02:
I'm also having similar problems during offboard control. GPS failsafe seems to be triggering, even though it is disabled.

saihv, 2017-08-10:
Apparently, the PIXHAWK commander module sets fail safe mode and sends a 'descend' command if a stream of offboard commands are not received with a 0.5s interval. Wonder if something on the AirSim's side is causing an offboard timeout.. I am going to look into the PIXHAWK NSH to see if there is any debug info printed.

lovettchris, 2017-08-10:
Offboard loss is normal if you are not telling AirSim to do anything.  Any pause between commands greater than about 50ms will cause offboard timeout to occur.  But I added a new COM_OBL_RC_ACT  value to latest PX4 version that should help with this:

param set COM_OBL_ACT 1
param set COM_OBL_RC_ACT 5

The COM_OBL_RC_ACT value 5 tells PX4 that when offboard timeout occurs, simple hover at the current location.  See [commander_params.c](https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander_params.c#L348) and [state_machine_helper](https://github.com/PX4/Firmware/blob/master/src/modules/commander/state_machine_helper.cpp#L838).

saihv, 2017-08-10:
@lovettchris Thanks for the response! I'm going to look into those parameters as well. The failsafe I mentioned was not because I didn't send any commands for a long time: it happens almost immediately after I send a command. For instance, if I say takeoff, the drone takes off and before it can reach its altitude, failsafe kicks in and it tries to land. And if I try takeoff again, it complains that takeoff altitude has already been reached (but the drone is now on the ground).

lovettchris, 2017-08-10:
Yeah, I've seen that problem too, last time I debugged it I found that the position estimation was drifting in the z-axis, so after takeoff z is drifting because altitude is drifting and that is what it ends up landing again.  I found that if I wait a bit for altitude to settle down then the take off is more reliable.

saihv, 2017-08-14:
This issue doesn't occur if I use PX4 firmware 1.6.3. Closing for now.

closed 2017-08-14 by saihv