microsoft/AirSim #396: [Request] Use other Keyboard Shortcuts for Compatibility with international Keyboards

EliteTUM, 2017-08-02:
I have similar Issues as mentioned in https://github.com/Microsoft/AirSim/issues/273 by lovettchris and https://github.com/Microsoft/AirSim/issues/50 by nacho-carnicero:

The Keyboard Shortcuts [ and ] do not work with non-english Keyboards, as they require two Keys to be pressed simultaniously:

[   is   Alt-Gr + 8   on German Layout
]   is   Alt-Gr + 9   on German Layout

For Spanish Layout I guess it seems to be similar according to nacho-carnicero.
Can you either change those Keys to regular available Keystrokes of e.g. Letters available on almost any Keyboard Layout?
Alternatively, does anyone have a Workaround on how to use the Shortcuts mentioned above on other layouts (like spanish, german, etc.)?

sytelus, 2017-08-02:
Thanks for pointing this out. I'll put this in our backlog. If you can't wait then its easy to change code: Below are the lines where you can specify the keys:

https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/CameraDirector.cpp#L79

So just change the keys to whatever you prefer and recompile.

sytelus, 2017-08-29:
These keyboard shortcuts are changed. Please use F1 to see new shortcuts. Also we have finally added simple_flight which eliminates need to set up PX4. Its also the default. So try out the latest version and use it with default settings (just delete settings.json if you have modified it).

closed 2017-08-29 by sytelus