microsoft/AirSim #397: Attempting to record image/training data causes UE to crash

Missyinvisible, 2017-08-02:
I am currently using the ComputerVision mode. Whenever I press the record button to capture image data Unreal crashes. 

Is anyone else having these kinds of issues?

sytelus, 2017-08-02:
Linux or Windows or Mac?

Missyinvisible, 2017-08-02:
Linux. I am actually using UE 4.15 since I found that to have the fewest other issues.

sytelus, 2017-08-03:
UE 4.15 will most definitely crash because of the bug I mentioned in my last comment. UE 4.16 has few workarounds. However, I think we should switch to UE 4.17 so those workarounds are not needed. I'd already tested with 4.17 and it was working well except for few issues like pressing end button caused UE to hang. I'm currently working on integrating firmware in AirSim but after that I plan to look at getting ironing out any issues in 4.17.

closed 2017-08-03 by sytelus