microsoft/AirSim #398: Issue when opening Airsim Block on Ubuntu 16.10

desireB, 2017-08-02:
I have installed QT Creator. My system is running Ubuntu 16.10 (System76 Oryx Pro, 32GB RAM, Nvidia GTX 1060 6GB).

I'm getting the following message : 
`
The project could not be compiled. Would you like to open it in an IDE to edit source code?
Running /home/desire/Tools/GitHubSrc/UnrealEngine/Engine/Binaries/DotNET/UnrealBuildTool.exe Blocks Development Linux -project="/home/desire/Tools/GitHubSrc/AirSim/Unreal/Environments/Blocks 4.16 - 2/Blocks.uproject" -editorrecompile -progress -NoHotReloadFromIDE
Setting swap interval to 'Immediate'
Running Mono...
~/Tools/GitHubSrc/UnrealEngine/Engine ~/Tools/GitHubSrc/UnrealEngine/Engine/Binaries/Linux
Creating makefile for Blocks (no existing makefile)
Performing full C++ include scan (no include cache file)
Using clang version '3.9.1' (string), 3 (major), 9 (minor), 1 (patch)
ERROR: Couldn't find module rules file for module 'AirSim'.
`

sytelus, 2017-08-02:
Did you do the ./setup.sh and ./build.sh for AirSim? Please make sure you have Plugins/AirSim folder under Blocks.

desireB, 2017-08-04:
Thank you for your answer. Yes, I put Plugin/Airsim and AirLib under BLocks.

I have now the following error : 
`
/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/AirSim/AirSim.generated.cpp:8:
In file included from ../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/AirSim/AirSim.generated.dep.h:15:
../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/PIPCamera.h:6:10: fatal error: 
      'controllers/VehicleCameraBase.hpp' file not found
#include "controllers/VehicleCameraBase.hpp"
         ^
In file included from /home/desire/Tools/GitHubSrc/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp:2:
/home/desire/Tools/GitHubSrc/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirBlueprintLib.cpp:11:10: fatal error: 
      'common/common_utils/Utils.hpp' file not found
#include "common/common_utils/Utils.hpp"
         ^
1 error generated.
1 error generated.
[2017.08.04-18.44.27:092][ 62][4/6] Compile Blocks.cpp
[2017.08.04-18.44.29:698][ 62]ERROR: UBT ERROR: Failed to produce item: /home/desire/Tools/GitHubSrc/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Binaries/Linux/libUE4Editor-AirSim.so
[2017.08.04-18.44.29:898][ 62]Total build time: 139.31 seconds (Local executor: 0.00 seconds)
`

I checked, and I do have the missing files : 
`
desire@desire-Oryx-Pro:~/Tools/GitHubSrc/AirSim/Unreal/Environments/Blocks$ find -iname VehicleCameraBase.hpp
./AirLib/include/controllers/VehicleCameraBase.hpp
desire@desire-Oryx-Pro:~/Tools/GitHubSrc/AirSim/Unreal/Environments/Blocks$ find -iname Utils.hpp
./AirLib/include/common/common_utils/Utils.hpp
`

Is that an issue that others have encountered and solved ?
Thank you.

closed 2017-08-14 by sytelus

desireB, 2017-08-14:
Hi,
Is there a fix for this issue ? I see that you closed it.
Thank you.