microsoft/AirSim #399: SITL Simulation using Joystick

Missyinvisible, 2017-08-03:
Hi guys!

I have followed these instructions to set up the SITL sim: https://github.com/Microsoft/AirSim/blob/master/docs/sitl.md

I then followed these instructions to use a joystick: https://github.com/Microsoft/AirSim/blob/master/docs/xbox_controller.md

And, in case anyone is wondering, I used these instructions to setup QGC: https://github.com/mavlink/qgroundcontrol  (An obvious mistake would be using the wrong version of Qt, but, I have 5.7.1)

I have followed the instructions exactly, however, when inside UE (after hitting the play button, of course) one of the messages displayed is "RC Controller on USB: Not detected" and I can't use my joysticks (or any of the buttons I preassigned in QGC) to control the drone. It looks like this:

[image]

I was wondering if anyone knows what is wrong, maybe I missed something, or if anyone has it working.

Thanks for all the help!

Robin-Wong, 2017-08-04:
I am using win10, SITL, XBOX controller, but QGroundControl-installer.exe download from web. I do not set as https://github.com/Microsoft/AirSim/blob/master/docs/xbox_controller.md, but only set the XBOX-button meening  in Joystick, Roll, Pitch, Yaw, Throttle, Arm.
Please try:
  "FpvVehicleName": "Pixhawk",
  "UsageScenario": "",
Not
  "FpvVehicleName": "SimpleFlight",
  "UsageScenario": "ComputerVision",
And must arm before control the drone, like set "A" button meening "arm" as me.

Missyinvisible, 2017-08-04:
Thanks for the suggestion, although I still appear to be having the same problem.

Missyinvisible, 2017-08-04:
I'm wondering if there is some way that you must link PX4 to QGroundControl?

Robin-Wong, 2017-08-06:
Have you set "UseSerial": false in settings.json?

Missyinvisible, 2017-08-07:
Yes, "UseSerial" has always been set to false in my settings.json file. Did you set up PX4? If so, how?

KhristopherK, 2017-08-08:
I am having the same issue. I have also configured the json file as per instruction and I can't get the sitl to talk to the sim.

Missyinvisible, 2017-08-08:
I was wondering as far as step #3 goes in the following link: https://github.com/Microsoft/AirSim/blob/master/docs/xbox_controller.md

How did you guys do this step? Like, where is MavLinkTest.exe located and where did you execute this command?

Robin-Wong, 2017-08-14:
I only set the #1 and #2 in https://github.com/Microsoft/AirSim/blob/master/docs/xbox_controller.md;
Just set up the PX4 and QGroundControl, when play in the UE4, there are log in PX4 and show connect in QGroundControl.

Missyinvisible, 2017-08-14:
That isn't working for me, unfortunately; you must have better luck! I see that a new version of the repo has just been released, so I am going to try that.

ChampDBG, 2017-08-16:
Hi, Do you check the option "Enable joystick input" in QGroundControl?
I checked the option and could fly with xbox one controller.
https://www.dropbox.com/s/7s0kn0sc2hja9lu/check.png?dl=0

Missyinvisible, 2017-08-16:
I finally got it working guys! In QGC under the Comm Links tab I added a new UDP Port. I used the port 14550, since I noticed in the settins.json file there is the following:

 "QgcPort": 14550,

This appears to have solved the problem for me, even when using an older version of AirSim and UE! :D

Missyinvisible, 2017-08-16:
I finally got it working guys! In QGC under the Comm Links tab I added a new UDP Port. I used the port 14550, since I noticed in the settings.json file there is the following:

"QgcPort": 14550,

This appears to have solved the problem for me, even when using an older version of AirSim and UE! :D

closed 2017-08-16 by Missyinvisible

hyybuaa, 2018-04-14:
@Robin-Wong how to set the XBOX-button meening in Joystick,I don't see it Joystick?

Robin-Wong, 2018-04-27:
I am using QGroundControl v3.2.4, there is no Joystick when the xbox controller does not connect
PC.
I am using win10, when the xbox controller connect PC, the Joystick will appear under second icon(wheel gear) of QGroundControl. 
Then we can calibrate the Attitude Controles, set the Enable joystick imput, set the Button actions as Arm,Position....