microsoft/AirSim #400: [HILS-HelloDrone] The Drone goes into failsafe mode after it takes off

sooyoungmoon, 2017-08-04:
Hello, I am testing the HelloDrone project in AirSim solution using HILS and have a problem in making the drone move. The current goal is to make my own project (similar to HelloDrone) to move drone by using moveToPosition command and save log (position, orientation and some other values)

Here is the screenshot.

[image]

Summary for my system settings.

1) PX4 version: 1.6.5
2) Airframe type (configured in QGC): HIL Quadcopter X
3) No radio setup since I don't use RC to control the drone.

When I connect the Pixhawk to the PC, the main LED is blinking red (fast) and doesn't change during the execution of AirSim. When I start the AirSim and HelloDrone, they work well until the taking-off.

However, the drone goes into failsafe mode right after it takes off and I cannot control the movement of the drone. (It immediately lands).

Could you give me any comment or tip to solve the problem?

Maybe I have to change some parameters related to the failsafe mode, but I have no idea about it.

Thank you in advance for your help.

saihv, 2017-08-09:
Can you try setting the parameters NAV_RCL_ACT and NAV_DLL_ACT to 0 (disabled) in QGroundControl? If the problem persists even after setting these to 0, then it's the same as the issue I'm having.

sooyoungmoon, 2017-08-10:
Thank you so much @saihv, I have set the parameters NAV_RCL_ACT (RC loss failsafe mode) and NAV_DLL_ACT (data link loss failsafe mode) to 'disabled' and could move the drone by using moveByVelocityZ() function. Still, the drone frequently goes into failsafe mode (ex. failsafe: no local position) and some runtime exceptions occur when moveByVelocityZ() is called (a timeout exception while RPC function is being executed) 
[image]
[image]
[image]

I hope your issue to be solved soon. Thank you!

closed 2017-11-14 by sytelus

vxgu86, 2019-07-02:
> Thank you so much @saihv, I have set the parameters NAV_RCL_ACT (RC loss failsafe mode) and NAV_DLL_ACT (data link loss failsafe mode) to 'disabled' and could move the drone by using [...]

did you solve your problem?
I am having the same problem even after two years!!