microsoft/AirSim #402: StereoImageGenerator produces dark Stere Images

EliteTUM, 2017-08-06:
I ran **runSteroImageGenerator()** from the Example-StereoImage-Generator from [here](https://github.com/Microsoft/AirSim/blob/master/Examples/main.cpp) using the [Blocks-Environment](https://github.com/Microsoft/AirSim/tree/master/Unreal/Environments/Blocks).

In general, the communication works: Once I run SimpleFlight in the BlocksEnvironment, Startup the StereoImage-Generator-Example in another executable as RPC-Client and the Connection is established properly and the left, right and disparity Image are saved (in the example for Random Positions).

When watching the Images being generated in the Unreal Editor, they look allright and the Scene seems to have proper lighting. Although, when taking a look at the Stereoimages, they all seem to be too dark. Outlines can only be seen as black silhouettes. Any idea why? Do you have the same Problem?

See the example Images attached (left and right Image of same timestep):
[image]
[image]

EliteTUM, 2017-08-06:
OK this is strange. Now that I see the uploaded Images, they appear to be mostly ok. Indeed, they even seem too bright. Looks like this is a Problem with the Default Win10-Picture-Viewer. When opening the Pictures above they rather look like this on my Screen (Screenshot taken using Snipping Tool):

[image]

EliteTUM, 2017-08-06:
I just tested different Settings for Quality (Engine Scalability Settings) and the results look the same. Additionally, when taking Pictures by clicking the record button they do look exactly the same: no textures recognizable, only the Background in somewhat blue Color, all objects dark/black.

sytelus, 2018-04-12:
Closing this as duplicate of https://github.com/Microsoft/AirSim/issues/352

closed 2018-04-12 by sytelus