microsoft/AirSim #405: How to make ultra sonic sensor

winszenYoung, 2017-08-08:
Hi , I have a few question about sensors.

I used gazebo, it was suportted many sensor(ultra sonic, 3d lidar, etc, optical flow)

I recently use airsim. I find sensor document about cam( depth, rgb) 

but, it is not have a another sensor, 

I want to use ultrasonic senosr or etc.

How do i use this sensor?

If it is not suportted, How do i make a ultrasonic senosr or etc?

sytelus, 2017-08-14:
Currently its not available out of the box so you will need to roll your own. Its not too hard to do. there is a SensorBase class which you can use as your base class and then use ray tracing to shoot few rays in predefined cone to get distances and then add some noise. There is a function `UAirBlueprintLib::GetObstacle` that you can use for ray tracing.

realdealneil, 2017-12-15:
I am trying to do this.  It's easy to set up the base class and make a class for a laser altimeter (which is all I want for the time being).  However, I don't yet know enough to figure out how to access the pawn's attitude and shoot a ray out to find where the laser would hit the ground.  

How do I access the pawn from the sensor class?  Here's what I'm trying right now (below is the getOutputInternal function from my LaserAltimeterSimple.hpp implementation):

Output getOutputInternal()
    {
        Output output;
        const GroundTruth& ground_truth = getGroundTruth();

        auto altitude = ground_truth.environment->getState().geo_point.altitude;
	auto position = ground_truth.environment->getState().position;
	auto attitude = ground_truth.kinematics->pose;

	output.laser_alt_m = altitude;      // TODO: fill in with true height above ground
	output.reading_raw_m = output.laser_alt_m;   // TODO: fill in with ray-traced value from vehicle pawn laser rangefinder

        return output;
    }

***********************************
I can access the pose of the aircraft (from which I can extract the attitude quaternion, cool).  However, UAirBlueprintLib::GetObstacle needs more stuff.  (Pointer to the right actor, start and end point of the trace, and then I need to handle the FHitResult).  

I'm thinking that the ground_truth structure should probably contain "terrain_height" and then the laser altimeter would just use this terrain_height and add some noise, etc.  If this is the right way to implement this, can someone who knows Unreal/Airsim better than me offer some advice on where to locate the function that does the ray tracing and give me some hints on how to call ::GetObstacle and interpret the result?

RVBldr, 2018-11-21:
Internal VSTF 7412

msb336, 2019-10-02:
Closing due to inactivity. Feel free to re-open if you are still actively working on this feature

closed 2019-10-02 by msb336