microsoft/AirSim #406: Can't find settings.json.

sj88eejqa, 2017-08-09:
I follow the instructions and still can't find settings.json in the documents/AirSim folder.
Can anyone help me?

viniciusguigo, 2017-08-10:
Hello,

What's exactly the error you are seeing? Please make sure you followed this doc: https://github.com/Microsoft/AirSim/blob/master/docs/settings.md

Thanks!

sj88eejqa, 2017-08-10:
My Documents\AirSim folder is empty.
Exactly, the path is 文件\AirSim(Chinese).
I don't know is that the reason or not.
My English is not good enough, hoping you can understand what I'm talking about.

lovettchris, 2017-08-10:
It could be a bug in the code related to globalization.  The code that creates the Settings file is as follows:

```
            void saveJSonFile(std::string fileName)
            {
                std::string path = getFullPath(fileName);
                std::ofstream s;
                common_utils::FileSystem::createTextFile(path, s);
                s << std::setw(2) << doc_ << std::endl;
            }
```

Can you see if you can get this snippet to work on your machine?  The core piece of this to check is the conversion of your User Documents folder to a std::string which happens here:

```
std::string FileSystem::getUserDocumentsFolder() {
    std::string path;
#ifdef _WIN32
    // Windows users can move the Documents folder to any location they want
    // SHGetFolderPath knows how to find it.
    wchar_t szPath[MAX_PATH];

    if (0 == SHGetFolderPath(NULL,
        CSIDL_MYDOCUMENTS | CSIDL_FLAG_CREATE,
        NULL,
        0,
        szPath))
    {
        std::wstring_convert<std::codecvt_utf8<wchar_t>, wchar_t> converter;
        path = converter.to_bytes(szPath);
    }

    // fall back in case SHGetFolderPath failed for some reason.
#endif
    if (path == "") {
        path = combine(getUserHomeFolder(), "Documents");
    }
    return ensureFolder(path);
}

```

So I'm wondering if the utf-8 conversion of your name is not working with std::ofstream.  I was told the VC++ std library supports utf-8, but perhaps this is wrong?

sj88eejqa, 2017-08-10:
I use std::ofstream to create a file, and it work.
My path is C:\Users\sj88e\OneDrive\文件\AirSim.
I feel confused.

lovettchris, 2017-08-10:
Can you post the code you just tested?

sj88eejqa, 2017-08-11:
ofstream outfile("C:/Users/sj88e/OneDrive/文件/AirSim/test.txt", ostream::binary);
	outfile << s;
	outfile.close();

I just make sure that the path can work.

lovettchris, 2017-08-11:
Thanks.  I did a test on my machine with that path and found that the problem lies with the conversion to utf-8 done by this code:

        std::wstring_convert<std::codecvt_utf8<wchar_t>, wchar_t> converter;
        path = converter.to_bytes(szPath);

You skipped that step which is why it worked for you.  So this test shows the interesting differences between Linux and Windows implementation of the std c++ library:

#include <string>
#include <fstream>
#include <codecvt>
#include <iostream>
#include <locale>
#include <cstring> 

```
int main()
{
#ifdef WIN32
    std::wstring unicodePath(L"C:\\Users\\sj88e\\OneDrive\\文件\\AirSim\\test.txt");
#else
    std::wstring unicodePath(L"/mnt/c/Users/sj88e/OneDrive/文件/AirSim/test.txt");
#endif
    std::wstring_convert<std::codecvt_utf8<wchar_t>, wchar_t> converter;
    std::string path = converter.to_bytes(unicodePath);
    std::ofstream outfile;
#ifdef WIN32
    std::wstring wide_path = converter.from_bytes(path);
    outfile.open(wide_path);
#else
    outfile.open(path);
#endif
    if (!outfile.is_open())
    {
        std::cout << "error opening file: " << std::strerror(errno) << std::endl;
        return 1;
    }

    outfile << "this is a test";
    outfile.close();
    return 0;
}

```

So I will have to do something like this in AirSim to solve this problem...

sj88eejqa, 2017-08-11:
Thank you, so how can I do now?

sj88eejqa, 2017-08-11:
Thanks, so what can I do now?

lovettchris, 2017-08-11:
So turns out AirSim was doing the right thing and it should be working.  So can you do one more test, can you try this function on your machine and print what it returns ?

```
#include <Windows.h>

std::string FileSystem::getUserDocumentsFolder() {
    std::string path;    // Windows users can move the Documents folder to any location they want
    // SHGetFolderPath knows how to find it.
    wchar_t szPath[MAX_PATH];

    if (0 == SHGetFolderPath(NULL,
        CSIDL_MYDOCUMENTS | CSIDL_FLAG_CREATE,
        NULL,
        0,
        szPath))
    {
        std::wstring_convert<std::codecvt_utf8<wchar_t>, wchar_t> converter;
        path = converter.to_bytes(szPath);
    }
    return path;
}
```

sj88eejqa, 2017-08-11:
I have another path "C:\Users\sj88e\Documents"(All of them is English).
Therefore, I just uninstall my OneDrive in windows 10, and assume that the Documents folders will build a AirSim folder like "C:\Users\sj88e\Documents\AirSim"(I rebuild AirSim), but the folder doesn't build.
The code above I test and it return "C:\Users\sj88e\Documents".

lovettchris, 2017-08-11:
so "C:\Users\sj88e\Documents" looks correct and AirSim should create an AirSim folder here, and the settings.json inside of that.

sj88eejqa, 2017-08-12:
I redo the instructions, but AirSim still not create an AirSim folder there.
[image]

sytelus, 2017-08-14:
Looks like this might be specific to your setup and so hard for us to reproduce or debug. If you are doing this on Windows with VC++ then its very easy to debug this. Just put break point at below location:

https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/SimMode/SimModeBase.cpp#L80

This is the place where we detect if settings file already exists and if not we create one. Let us know what you find.

sj88eejqa, 2017-08-14:
[image]

sj88eejqa, 2017-08-14:
Well, I don't know why but AirSim folder is created.
Thanks for helping.

closed 2017-08-14 by sj88eejqa

Kjell-K, 2017-08-25:
@lovettchris @sytelus I have the same issue as @sj88eejqa. Unfortunately it does not resolve. 
When I compile SimModeBase.cpp my droneShell is waiting for connection. My Documents/ folder is empty. I just want to change to computer vision mode.

[image]