microsoft/AirSim #408: Crash on simulation restart

KeeganVG, 2017-08-10:
I could have sworn I saw that this was a known bug somewhere, however has anyone come up with a patch for it? I'm trying to use reinforcement learning, and having the ability to reset the simulation is a must.

It doesn't have to be through the backspace, anything that would let me restart the simulation programmatically would work!

Any help is appreciated!

sytelus, 2017-08-14:
You probably don't want to do RL on PX4. We are developing new firmware called simple_flight that is directly integrated with arisim and would have reset ability as you described above. Right now I have most things working for simple_flight and hope to get it out by end of week.

KeeganVG, 2017-08-14:
I appreciate the advice! Is simple_flight SITL based or does it still use the PX4s simulation mode?

KeeganVG, 2017-08-29:
With the release of simple_flight is there any plans to fix the crashing issue? I just pulled the recent changes and am still experiencing the crashes?

My configuration may be wrong, any help is greatly appreciated! :D

sytelus, 2017-09-10:
If you have everything latest, try deleting your settings.json. After rebuilding, make sure you run update_from_git.bat in Blocks project so previous intermediate files are cleaned up and new one gets copied. This will cause everything to default and simple-flight would be used. If you still experience crash, please submit exact error or look under the Saved folder for logs.

closed 2017-11-14 by sytelus