microsoft/AirSim #409: Replace common mavlink messge set with ardupilotmega mavlink message set

auturgy, 2017-08-11:
The ardupilotmega mavlink dialect is a superset: it includes common, uavionix and it's own custom messages.  Replacing mavlink common with ardupilotmega opens the use of AirSim to a user base that is an order of magnitude (or more) larger than restricting to common, whilst maintaining complete compatibility with the px4 codebase.
For example QGroundControl uses the ardupilotmega dialect for exactly the reason I have articulated above.
Noting the issues you have had with px4, it makes sense to do this!

sytelus, 2017-08-14:
That's great idea. Unfortunately I'm not familiar with ardupilotmega's version of MavLink. However if you have any experience with it, we had love to get pull request :).

auturgy, 2017-08-14:
Lol. Ok. I can do that, but it's as simple as switching common.xml for Ardupilotmega.xml. Good to know you're open to it though 

Regards,

James

sytelus, 2017-11-14:
@lovettchris might be interested in this.

closed 2017-11-14 by sytelus