microsoft/AirSim #410: UdpClientPort socket could not connect to remote host at (remote ip):14560, error: 10051

ZixingLin, 2017-08-11:
Hello everyone,

I recently installed AirSim on my Unreal Engine 4.16.2 and ran PX4 SITL in VirtualBox Ubuntu. Reading on documentations I've also editted my settings.json to ensure the IP addresses are set correctly: LocalHostIP is set to the IP address of my host machine (Windows 7 machine) and SitlIP, UdpIP are set to my virtual machine's IP. Both these addresses are able to ping each other and are on the same network, below are snippets of the settings.json with regards to IP addresses:

"LocalHostIp": "170.95.67.38"

"SitlIp": "170.95.67.99",
"SitlPort": 14556,
"UdpIp": "170.95.67.99",
"UdpPort": 14560,
"UseSerial": false

However, after getting PX4 SITL to run on my virtual machine and running AirSim in UE4, I am constantly faced with this error: UdpClientPort socket could not connect to remote host at 170.95.67.99:14560, error: 10051
I'm quite baffled and I was unsure what could be causing this error, I have attached screenshots below which indicates the error, as well as the PX4 SITL terminal.

Any help would be appreciated!
[image]
[image]

ZixingLin, 2017-08-15:
Update: Problem still persists - however this problem seems fairly exclusive as most similar problems are resolved due to having wrong settings.json IP addresses. Hence it has lead me to think that there is a potential issue with the proxy that is installed in my company's network, could that have caused a problem in AirSim connecting to the VM's PX4 SITL?

Any form of input is very much appreciated. Thank you.

closed 2017-08-29 by ZixingLin

MrChill, 2017-12-12:
I am facing the same issue? What was the solution?
Regards