microsoft/AirSim #413: No mavlink tab or Airsim specific properties in Unreal Engine

wuwushrek, 2017-08-11:
Hello , 

Thanks for the amazing job you are doing here guys.
I'm trying to flight using PX4 SITL  multiples vehicles and as I read from this topic  #92 , replicating FlyingPawn and PIPCamera assets are the way to do it. The thing is in my Unreal Environment I don't find the mavlink tab to associate FlyingPawn with as it has been showed in the [Unreal Airsim Setup video](https://youtu.be/1oY8Qu5maQQ).
[image]

Thanks in advance for the help.

sytelus, 2017-08-14:
Above properties existed because previously we used something called "Blueprint" feature of Unreal engine. Unfortunately there are several downsides using Blueprints in Unreal including the fact that they get stored in binary opaque files making it very difficult to see diffs and prevent inadvertent changes. Long story short, we ended up moving most of the code from Blueprints to C++.

While reviewing code, I realized that multi-vehicle scenario was broken. So I just made few fixes however I haven't got chance to test it. If you had like to go for test ride, get latest and please take a look at this doc: https://github.com/Microsoft/AirSim/blob/master/docs/multi_vehicle.md

saihv, 2017-08-14:
@sytelus Nice to see that multi vehicle is working! Does the API provide access to the images coming from the various vehicles as well?

wuwushrek, 2017-08-14:
@sytelus thanks for the modifications you made.

I tested it this morning and wasn't able to set multiple vehicles. With one vehicle it works fine if you drop an instance of BP_FlyingPawn and set a name according to the settings.json and press play button.

But when I add a second quad and set his own name and then press play, UE4Editor just turn black , no message error. It just freezes. I can just see the scene background but can't access anything on UE4Editor. The only way to close it is to quit force on Ubuntu. Even Ctrl-C doesn't work.

saihv, 2017-08-15:
Just tried it out myself: multi-vehicle scenario crashes in Windows as well: throws exception at a "return rotors_.at(index);" line in Multirotor.hpp when 'Play' is clicked after placing multiple flying pawn instances. Happens with both Pixhawk and SimpleFlight types.

sytelus, 2018-04-12:
We are re-working multi-agent scenario. Meanwhile use this thread for discussions which is very similar: https://github.com/Microsoft/AirSim/issues/431

closed 2018-04-12 by sytelus