microsoft/AirSim #415: takeoff error

lxgen, 2017-08-14:
I use the lastest airsim build on ue4.16. When I start the PX4 in SITL and ue4 project, it always shows the following text on the px4 terminal. And when I type "pos" in DroneShell, the "z" value is in great change from 0 to 20. I also can't takeoff the drone. How can I fix the problem? Thank you.
[image]

Robin-Wong, 2017-08-14:
For me on win10, if I use:
  "FpvVehicleName": "Pixhawk",
  "UsageScenario": "",
I must take the arm using xbox_controller when connected with QGroundControl through, then I can takeoff the drone.

sytelus, 2017-08-14:
Which PX4 release are you using? Also are you saying typing pos in PX4 terminal (not DroneShell) gives you varying values between 0 to 30?

lxgen, 2017-08-14:
@sytelus I use 1.6.0 Release RC1 of PX4 and have successfully run airsim built on ue4.14 in SITL mode with this version of PX4.  However it doesn't work on the lastest airsim built on ue4.16. And I type pos in DroneShell, just like the following picture. 
[image]

sytelus, 2017-08-14:
I can reproduce this in 1.5.5 as well. Looks like above are the estimates done in PX4 so the estimator in PX4 is basically grocery wrong. I'm not sure why... If you want to get ground truth position, then you can use RPC API call `simGetPose()` (see https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md). Or you can press "R" key to see live report of state.

Hopefully, soon we will move on to simple_flight firmware as default and won't have these PX4 related issues. If you are familiar with PX4 debugging, please do feel free to provide any help!

lxgen, 2017-08-15:
@sytelus Sorry to ask you again. With this problem, I can't call APIs to control the drone successfully in SITL, such as arm, takeoff, MoveToz and so on. Is there any other successful way to call APIs to control the drone in SITL?

sytelus, 2017-08-29:
We have finally added simple_flight which eliminates need to set up PX4. Its also the default. So try out the latest version and use it with default settings (just delete settings.json if you have modified it). You should then be able to use APIs. Again no extra HITL/SITL setup needed.

AlexandreBorowczyk, 2017-10-26:
It seems I've got the same issue. Using ue4.17 and px4 1.5.5. In QgroundControl, I can also see the altitude change greatly and rapidly. on the UE side there are warning:
`MavLinkConnectionImpl: Error handling message 0 on connection 'hil', details: MavLinkConnectionImpl: Error sending message on connection 'Proxy to: LogViewerOut at 127.0.0.1:14389', details: UdpClientPort socket send failed with error: 111
MavLinkConnectionImpl: Error handling message 83 on connection 'sitl', details: MavLinkConnectionImpl: Error sending message on connection 'Proxy to: QGC at 127.0.0.1:14550', details: UdpClientPort socket send failed with error: 111
MavLinkConnectionImpl: Error handling message 230 on connection 'sitl', details: MavLinkConnectionImpl: Error sending message on connection 'Proxy to: LogViewer at 127.0.0.1:14388', details: UdpClientPort socket send failed with error: 111
`

lovettchris, 2017-12-09:
This might be fixed now with [my new pull request](https://github.com/Microsoft/AirSim/pull/670).

lovettchris, 2017-12-09:
merged.

closed 2017-12-09 by lovettchris