microsoft/AirSim #416: DroneShell error

akshaysw, 2017-08-14:
[image]
can anyone tell me why am i getting this error

akshaysw, 2017-08-14:
[image]
this is my setting file.

akshaysw, 2017-08-14:
I thought i made some mistake while installing airsim because keyboard keys were not working, so I deleted the entire airsim directory and again cloned it from git and followed this instructions [https://github.com/Microsoft/AirSim/blob/master/docs/build.md](url) then I created new project made following changes in settings.json as shown in the following picture. now in computer vision mode is active and keyboard keys are also working but when I'm trying to use DroneShell it's giving different error as shown in the following picture. @sytelus thanks in advance
[image]
[image]
[image]

sytelus, 2017-08-14:
Commands that require physics are not supported in CV mode. I guess we should document this better. In CV mode, you don't have drone, you just have cameras. It is supposed to be used with APIs simSetPose to set pose for camera and simGetImages to get images. Nothing else is expected to be functional.

We are soon planning to release simple_flight firmware which will have full drone functionality but without needing PX4.

closed 2017-08-14 by sytelus