microsoft/AirSim #418: How to show a planned trajectory in the Simulator?

Mayankm96, 2017-08-16:
I am trying to implement a trajectory planning algorithm using the simulator. While I am interfacing with the simulator sing the ROS wrappers suggested in [#269](https://github.com/Microsoft/AirSim/issues/269), and able to get a goal point, I would like to show the path that is being planned in the simulator itself; something similar to when you press `T` the pink line is made showing it's trajectory.

Would love to know if there is a way to do this. Thank you!

sytelus, 2017-08-29:
I'm not sure what you mean... Do you mean simply draw a line to the goal point from current point? The actual flight path is determined dynamically and is constantly corrected for any deviations.

Mayankm96, 2017-09-27:
I apologize for replying so late on this thread. Well it could be as simple as drawing a line through the waypoints that are being set. 

What I really wanted to show was that once I had the planned trajectory I could visualize it on the AirSim directly, similar to the way one can in rviz. Nonetheless, I am using rviz for this now so it's not required. 

I would love to get to know when the ROS wrappers would be officially released for the AirSim. It would be handy to have those now. :)

closed 2017-09-27 by Mayankm96

Hoshea7, 2019-06-11:
> > I'm not sure what you mean... Do you mean simply draw a line to the goal point from current point? The actual flight path is determined dynamically and is constantly corrected for any deviations.
> 
> I want to draw a line to the goal point...how can I make it...

Can you make it now ? I also want to do this on a drone trajectory.

DavidLSmyth, 2019-06-11:
I think issue #1393 outlines what you are looking for, I added this to the API in a fork. I've broken things since then but the changes you need to make to the source code are here: https://github.com/DavidLSmyth/AirSim/commits?author=DavidLSmyth. If you don't need some of the newer functionality, you could clone/fork my repo from commit d41137da39b2e82db769a7e9b03d7bcefc04d3ff