microsoft/AirSim #419: I have some questions about simple_flight

yonggi11, 2017-08-16:
First of all, thanks to sytelus who makes simple_flight. 

I want to use simple_flight to train Reinforcement Learning. but have some problems

Q1. "requestApicontrol was successful" is done, while I push joystick buttons. is there another way to     
       print "requestApicontrol was successful" in unreal engine? otherwise, RPC command is rejected

Q2. when I use ps4 controller, unrealengine show following message.
                                            "RC controller on USB : Detected" 
                                            "Joystick (T, R,P, Y) x.x x.x x.x x.x" 
                                            "RC Mode : Angle (Rate)"
      but I can't use manual control using joystick. vehicle is not moving. so how to use manual control     
      using joystick?

Q3. I transmit RPC command using DroneShell. but vehicle is not moving along the x, y axis. moving 
      along z axis. 
      Example, 1. it works when using commandVelocity 0, 0, -5
                     2. it works along z axis, but not x axis when using commandVelocity 5, 0, -5
                     3. it works along z axis, but not y axis when using commandVelocity 0, 5, -5
                     4. it works along z axis, but not x axis ,y axis when using commandVelocity 5, 5, -5
                     5. Only the propeller in the diagonal line is rotated, That is, only two of the four are rotated.
      
      what is the problem??

Q4. I heard that you are making a reset button in simple_flight. When will it be finished? and Can I start 
       at an arbitrary starting point? I think it would be better to start at an arbitrary starting point.  

I want you to answer quickly. Because I am interested in simple_flight. Thank you once again for making simple_flight.

sytelus, 2017-08-29:
Please try out the latest version. The simple_flight is now official and default. I haven't used PS4 controller but XBox controller seems to work fine. DroneShell also works fine in latest version. To set your starting point, move the "Player Start" component in Unreal environment. The reset API is not yet done... Probably in couple of weeks...

Kjell-K, 2017-09-14:
@sytelus The reset API you were mentioning, is that `simSetPose` now? Cause can not get it running for simple_flight.
For example,
`client.simSetPose(Vector3r(-10,10,-10),Quaternionr(1,0,0,0))`
, does not work in simple_flight.

sytelus, 2017-09-19:
I found the reason. The Quaternionr class didn't had constructor in Python client. I just [fixed it](https://github.com/Microsoft/AirSim/commit/1fe5e403db32c2adac21a4e920cea75b6118b903) and now it works. Here's the [sample code ](https://github.com/Microsoft/AirSim/blob/master/PythonClient/cv_mode.py)I tested.

closed 2017-09-19 by sytelus

Kjell-K, 2017-09-19:
Awesome! Thanks!