microsoft/AirSim #421: Arming the Drone via Simple_Flight

ZixingLin, 2017-08-16:
Hi everyone,

It seems that SimpleFlight has been updated and working, however there isn't any documentations on it and I'm unsure how exactly to use it. I have changed the DefaultVehicleConfig in settings.json to SimpleFlight but when I tried to arm the drone via DroneShell - the following error is given: 

RPC Error: rpclib: function 'armDisarm' (called with 1 arg(s)) threw an exception. The exception contained this information: Vehicle cannot be armed via API because API has not been given control.

If anyone has a fix to this do let me know, and I'd appreciate any additional information on using Simple Flight as I wish to use it to run Python scripts to AirSim without PX4.

Strupi11, 2017-08-16:
I guess for that you have to run RequestControl command. 

But in order to get control, you need to allow offboard control using an RC switch. Hence with SimpleFlight you ned to get a joystick working. And that's exactly where I fail. It won't detect my connected XBox360 wireless controller, no matter what ID. 

It works fine in the Unreal viewport editor, but it is not detected when running Airsim. I also did a quick test according to this code: https://katyscode.wordpress.com/2013/08/30/xinput-tutorial-part-1-adding-gamepad-support-to-your-windows-game/. Here connection to the gamepad works fine as well!

ZixingLin, 2017-08-17:
Thank you for your reply Strupi! I'll get to testing with my xbox controller as well and will update you if I manage to get it working in AirSim.

ZixingLin, 2017-08-17:
Hi Strupi, I have connected my controller and exactly as you say, I'm facing the same issue - Controller works in viewport editor but not when I run AirSim. By any chance did you get yours to work?

Strupi11, 2017-08-17:
No, I was not yet able to get it running. 
The problem is in SimJoyStick.cpp, getJoyStickState() where XInputGetState() just doesn't see a connected gamepad. Unfortunately I don't know why and how I could debug that.

closed 2017-08-29 by ZixingLin