microsoft/AirSim #424: Free Unreal Engine Environments

robotvin, 2017-08-19:
Please provide me links where i can get free unreal engine environments.
Where i can get the environment shown in Manual Flight Demo section.

Timebutt, 2017-08-21:
The excellent [step-by-step tutorial](https://github.com/Microsoft/AirSim/blob/master/docs/unreal_custenv.md) describes in detail how you can get AirSim running with the free LandscapeMountains environment.

themmer, 2017-11-13:
@robotvin Looks like Epic Zen Garden is free as well. You should pry close this issue.

sytelus, 2017-11-14:
Look in to the Learn tab of Epic Game Launcher. You might find environments that you can add to project or full ready made projects (in which case delete any startup matinee stuff).

Also, here's tutorial on how to import 3rd party environments from TurboSquid.com or cgitrader.com: https://www.youtube.com/watch?v=y09VbdQWvQY

[image]

closed 2017-11-14 by sytelus

chris-aeviator, 2019-05-05:
Where can I get the environments supplied in the Windows (& now exp. ubuntu builds) e.g. the city environment?