microsoft/AirSim #428: Line	Suppression State Error	LNK1104	cannot open file 'rpc.lib'	HelloDrone	F:\UAVHIDL\Airsim\Airsim\HelloDrone\LINK	1

icydust, 2017-08-21:
when i build the airsim.sln the rpc,lib ,How can i solve this problem?
Waiting for an answer online
sincerely！

Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK1104	cannot open file 'rpc.lib'	HelloDrone	F:\UAVHIDL\Airsim\Airsim\HelloDrone\LINK	1	
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK1104	cannot open file 'rpc.lib'	Examples	F:\UAVHIDL\Airsim\Airsim\Examples\LINK	1	
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK1104	cannot open file 'rpc.lib'	DroneShell	F:\UAVHIDL\Airsim\Airsim\DroneShell\LINK	1	
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK1104	cannot open file 'rpc.lib'	DroneServer	F:\UAVHIDL\Airsim\Airsim\DroneServer\LINK	1	
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK1104	cannot open file 'F:\UAVHIDL\Airsim\Airsim\AirLib\lib\x64\Debug\AirLib.lib'	AirLibUnitTests	F:\UAVHIDL\Airsim\Airsim\AirLibUnitTests\LINK	1	
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK1181	cannot open input file 'rpc.lib'	AirLib	F:\UAVHIDL\Airsim\Airsim\AirLib\LINK	1	

[image]
@sverrejoh

sytelus, 2017-08-29:
Please make sure you are using the latest version have followed the [build doc](https://github.com/Microsoft/AirSim/blob/master/docs/build_windows.md) properly.

closed 2017-08-29 by sytelus

icydust, 2017-08-29:
Thank you for your help，i try to dowload the  file with the git clone ,and rebuild it , now i can rebuild it by vs2015,but when i use the "developer command prompt for vs2015,it have an other error.\
like this :
[image]
[image]
[image]

 @sytelus

icydust, 2017-08-29:
when i launch myproject with unreal it have this error:
[image]

How can i solve this problem?
@sytelus