microsoft/AirSim #429: takeoff command is not working

akshaysw, 2017-08-21:
hello, I am trying to use drone shell to control drone but I am getting following error. what should I do?

RPC Error: rpclib: function 'takeoff' (called with 1 arg(s)) threw an exception. The exception contained this information: TakeOff command rejected by drone.
[image]

sytelus, 2017-08-29:
Looks like PX4 is not allowing you to takeoff. This could be because of many reasons such as RC switches are not down or you have buggy release of PX4.

akshaysw, 2017-08-31:
hello @sytelus i have followed all the step you mentioned above but now i am getting following error, my setting.json is attached below
[image]
[image]
[image]

sytelus, 2017-09-01:
Can you see the value of message parameter? You can just continue or comment out that statement. It's just there to let you know there was some error message from flight controller. Also please make sure to use the latest version.  We also have binary release now that you can just simply download and run: [See docs](https://github.com/Microsoft/AirSim/blob/master/docs/use_precompiled.md) for more info.

akshaysw, 2017-09-02:
hello @sytelus yes binary is working perfectly fine and API's are also working thanks a lot though i could not figure out what was the problem before. can you please tell me how did you make binary?

sytelus, 2017-09-09:
Binary file is using simple_flight as flight controller instead of PX4. I believe the problem was in PX4 setup before.

closed 2017-09-18 by akshaysw