microsoft/AirSim #430: A few error using Px4

winszenYoung, 2017-08-21:
Hi I'm using airsim with px4.

It's errotic action in airsim.

I try to real px4. 

I follow this ref [Prerequisites](https://github.com/Microsoft/AirSim/blob/master/docs/prereq.md) [Walkthrough Demo Video](https://www.youtube.com/watch?v=HNWdYrtw3f0&feature=youtu.be)

RC control works quite good. but Droneshell, and QGC control is not work.

It's not work by altitude error.

I record video,  you can see this link [ Px4 with rc](https://www.youtube.com/watch?v=5mSF0JjzOoE)
[Px4 with Droneshell](https://www.youtube.com/watch?v=vOMi_9CClB4)

Environment
 Os  win10
 px4 version : v1.5.4, frame: hitl config x           qgc v 3.2.2
 unreal engine v4.16.3        airsim commit: d7cd82bd74d2547d1b470235164f20ca992cdbe8

And I try to Sitl 
I follow this ref [sitl](https://github.com/Microsoft/AirSim/blob/master/docs/sitl.md)

I record video, It same error [ video link](https://www.youtube.com/watch?v=1OCleMM-CeI)

I think this error occur by ground and drone collison.

Drone is flying altitude is good.
Drone and ground are collison. Altitude continue down. 

How do I fix it?

ps. I try to using rosflight with droneshell, It works not.( I try to take off or etc)

ps2.
I try to  many environment, it seems same error.
Os : win10, ubuntu 16.04
Pixhawk version: v1.5.5 v1.5.4 , v1.6.5
Unreal engine: v.4.16.3  
Airsim commit: 
2a59fe76353056a26674c3fdfd790fc910f1de85
d7cd82bd74d2547d1b470235164f20ca992cdbe8
08657eaed70e001d0495bfa01e2b14861d6148b6

sytelus, 2017-08-29:
I'll try to reproduce this. Meanwhile we have released new version. We have finally added also added simple_flight which eliminates need to set up PX4. Its also the default. So try out the latest version and use it with default settings (just delete settings.json if you have modified it).

closed 2017-11-14 by sytelus

machenxiang, 2019-03-18:
@winszenYoung hi winszenYoung wehave the same question. Have you solved this problem? If yes can you tell me how to solve this issue?