microsoft/AirSim #431: Can't use BP_CameraDirector to chase the drone

jimchenhub, 2017-08-21:
I always use 'Flay with me' View. But when I fly in the forest, the drone will be shaded from the trees and and leaves. So I want to use a fixed view to chase the drone as mentioned in [youtube setup video](https://www.youtube.com/watch?v=1oY8Qu5maQQ&feature=youtu.be) at the end. 

I first made sure there is no `Player Start` in the environment, and added the a BP_FlyingPawn, a BP_PIPCamera and a BP_CameraDirector.

But in the `Details` of BP_CameraDirector as shown below, there is no `target pawn` and `external cameras` .

[image]

Is there a way to add a chasing camera in this latest version?

sytelus, 2017-08-29:
These properties were accidentally made private in recent refactoring. We will restore them soon.

edithcomas, 2017-09-30:
Has this feature been restored? Have you found a workaround for chasing in the meantime?

GitWyd, 2017-10-04:
@sytelus Could you please let us know which properties they were, and which files are affected? Then we could temporarily fix the code until they are restore.

HarrisSym, 2018-01-31:
@sytelus is there an update to this problem?

jimchenhub, 2018-01-31:
Hi @HarrisSym, now it can use manual view mode during the flight. Just click M to toggle it. You can press F1 for more detail.

HarrisSym, 2018-01-31:
@jimchenhub  I  was refferring to  the issue with the properties of the BP_CameraDirector Blueprint.

jonyMarino, 2020-11-06:
related to #3050 use case

zimmy87, 2021-03-12:
This is an old issue. Now things can be configured through settings.json. Closing until someone is interested in having this again.

closed 2021-03-12 by zimmy87