microsoft/AirSim #432: Python Client doesn't work

fshamshirdar, 2017-08-21:
I've got some errors on the latest commit including: sequence of freefall detection on SITL, crashing by running DroneShell navigation commands (like MoveToZ), safefail landing right after taking off, and more importantly for my case Python Client:

\>python takeoff.py
Waiting for home GPS location to be set...
Traceback (most recent call last):
  File "takeoff.py", line 10, in <module>
    home = client.getHomePoint()
  File "AirSim\PythonClient\PythonClient.py", line 65, in getHomePoint
    return self.client.call('getHomePoint')
  File "C:\Python27\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "C:\Python27\lib\site-packages\msgpackrpc\future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: server could not find function 'getHomePoint' with argument count 0.

ChampDBG, 2017-08-29:
I've the same errors about "takeoff.py", is there any suggestion ?
I will appreciate your help with this situation.

sytelus, 2017-08-29:
Duplicate https://github.com/Microsoft/AirSim/issues/441

closed 2017-08-29 by sytelus